openvla/openvla #115: Minimum information required for OpenVLA Fine-Tuning

shifsa, 2024-09-19:
Let me confirm this, as I could not find the information after checking various codes. 
Is the minimum information required for OpenVLA fine-tuning the image, state (current robot posture information), action (movement to the next step), and language instruction for each step?

jonathonsather-bc, 2024-09-23:
@shifsa Minimum is image, action, and instruction. Other state is not used by default

esychoi, 2024-09-26:
Is it possible to make the model use the state information ? like the robot position of some other sensors.

ClousTom, 2024-09-26:
What is meant by action? Joint and EEF positions or joint velocities?

jonathonsather-bc, 2024-09-26:
@ClousTom Action is a 7 dof vector corresponding to EEF position (6) + gripper state (1), and I think the specific representation is dataset-specific. I know in the bridge dataset, actions are `(Δx, Δy, Δz, Δroll, Δpitch, Δyaw, gripper open/close)`, where translation and rotation are applied to the end effector in the global frame, as done [here.](https://github.com/rail-berkeley/bridge_data_robot/blob/b841131ecd512bafb303075bd8f8b677e0bf9f1f/widowx_envs/widowx_envs/base/robot_base_env.py#L184). For example, if your end-effector pose is `(x, y, z, roll, pitch, yaw)`, applying the action yields `(x + Δx, y + Δy, z + Δz, *to_rpy(to_rotation_matrix(Δroll, Δpitch, Δyaw) @ to_rotation_matrix(roll, pitch, yaw)))`.

jonathonsather-bc, 2024-09-26:
@esychoi You could try adding it to the prompt, as in https://github.com/openvla/openvla/issues/64. Looks like you may need a large dataset for this

esychoi, 2024-09-27:
@jonathonsather-bc Thank you !!

kpertsch, 2024-10-08:
Thanks for covering the discussion here -- agreed with all that has been said. Adding proprio is not something we have experimented with, but should be easy through the language prompt!

closed 2024-10-08 by kpertsch

ClousTom, 2024-10-21:
Hello. Sorry if I open again this thread. I have registered my dataset in this way:

    "baxter_dflacon": {  # My dataset from Baxter (grasp the detergent flacon)
        "image_obs_keys": {"primary": "image", "secondary": "head_image", "wrist": None},
        "depth_obs_keys": {"primary": None, "secondary": None, "wrist": None},
        "state_obs_keys": ["state"],
        "state_encoding": StateEncoding.JOINT,
        "action_encoding": ActionEncoding.JOINT_POS,
    }

as my representation is: 
- joint angles (7) + gripper state (1), for state
- joint delta position (7) + gripper state (1), for action 

but running the finetuning command I receive this error:
10/21 [19:06:34] WARNING  | >> [*] Skipping `baxter_dflacon`  materialize.py:129
                          due to Error: Cannot load                             
                          `baxter_dflacon`; only EEF_POS &                      
                          EEF_R6 actions supported!

why? is  "action_encoding": ActionEncoding.JOINT_POS not supported?

songlin, 2024-10-30:
> @ClousTom Action is a 7 dof vector corresponding to EEF position (6) + gripper state (1), and I think the representation is dataset-specific. I know in the bridge dataset, actions are `(Δx, Δy, Δz, Δroll, Δpitch, Δyaw, gripper open/close)`, where translation and rotation are applied to the end effector in the global frame, as done [here.](https://github.com/rail-berkeley/bridge_data_robot/blob/b841131ecd512bafb303075bd8f8b677e0bf9f1f/widowx_envs/widowx_envs/base/robot_base_env.py#L184). For example, if your end-effector pose is `(x, y, z, roll, pitch, yaw)`, applying the action yields `(x + Δx, y + Δy, z + Δz, *to_rpy(to_rotation_matrix(Δroll, Δpitch, Δyaw) @ to_rotation_matrix(roll, pitch, yaw)))`.

I think the $\Delta$ rpy should be right-multiplication?