openvla/openvla #118: Evaluating OpenVLA scratch on LIBERO

wu-yutong-525, 2024-09-22:
Hi there, 

  Great work! May I kindly ask is there way to test not finetuned version OpenVLA's performance on LIBERO benchmark? I try to directly revise the checkpoint name to openvla/openvla-7b but I get the following error:
```
"/home/yutong/openvla/experiments/robot/libero/run_libero_eval.py", line 288, in <module>
    eval_libero()
  File "/home/yutong/miniconda3/envs/openvla/lib/python3.10/site-packages/draccus/argparsing.py", line 203, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/yutong/openvla/experiments/robot/libero/run_libero_eval.py", line 114, in eval_libero
    assert cfg.unnorm_key in model.norm_stats, f"Action un-norm key {cfg.unnorm_key} not found in VLA norm_stats!"
AssertionError: Action un-norm key libero_spatial not found in VLA norm_stats!
```

jis3613, 2024-10-07:
An OpenVLA model requires a "unnorm_key" for the data it is trying to inference. During training, OpenVLA modle calculates the statistics of the dataset it is being trained on and saves it. 
Therefore, the pre-trained OpenVLA model that is not fine-tuned on the LIBERO dataset does not have the unnorm_key for the Libero dataset, so the above error raises.
You can tweak the code to explicitly make the model to use the unnorm_key of the other dataset that the pre-trained OpenVLA has already been trained on, but it would not fit to the Libero dataset, hence model is likely to fail.

kpertsch, 2024-10-08:
Thanks for the answer above -- I agree with all that was said. Technically you *can* evaluate OpenVLA without LIBERO finetuning on LIBERO, you can just choose the unnorm key from a different dataset from the pre-training mix, like any of the Franka robot datasets.

**However**, it's very likely that the model completely fails since it's not conditioned on the action space definition of the LIBERO environment and only implicitly learns it **when finetuning on LIBERO**. Thus, the finetuning stage is crucial for good performance on a robot setup not seen during pre-training of OpenVLA.

closed 2024-10-08 by kpertsch