openvla/openvla #119: How to register and transform a custom dataset with your data loader

ClousTom, 2024-09-23:
"For option (a), after you converted your dataset to RLDS, you need to register it with our data loader, by registering a dataset config [here](https://github.com/openvla/openvla/blob/main/prismatic/vla/datasets/rlds/oxe/configs.py#L54) and a dataset transform function [here](https://github.com/openvla/openvla/blob/main/prismatic/vla/datasets/rlds/oxe/transforms.py#L828)."

This is the sentence explaining what I have to do in the section "Fine-Tuning OpenVLA via LoRA". Now, I tried to convert to RLDS format the example dataset of the repo "RLDS Dataset Conversion".

The numpy files of the starting example dataset was:
Element 0:
  image:
    Data type: uint8
    Dimensions: (64, 64, 3)
    Example values: [247 104 170  44  53 130 222  40  38 193 208 140 186 194 229] ...
  wrist_image:
    Data type: uint8
    Dimensions: (64, 64, 3)
    Example values: [201  84  86   1 194  23 211 188  61 100 181 142 254  25 252] ...
  state:
    Data type: float32
    Dimensions: (10,)
    Example values: [0.10125669 0.09073878 0.09817205 0.89327097 0.8034896  0.99834484
 0.7588733  0.9112549  0.737019   0.77097785] ...
  action:
    Data type: float32
    Dimensions: (10,)
    Example values: [0.38481426 0.5523757  0.14422442 0.9207629  0.79871047 0.7309072
 0.01864261 0.96003103 0.17583813 0.7490036 ] ...
  language_instruction:
    Data type: <class 'str'>
    Value: dummy instruction

Element 1:
...
    
After the conversion, I have four file in "example_dataset/1.0.0/:
    - dataset_info.json
    - features.json
    - example_dataset-train.tfrecord-00000-of-00001
    - example_dataset-val.tfrecord-00000-of-00001

I also tried to run the procedure in the section "Add Transform for Target Spec" of that repo and it works.

Now, I have to register and transform it with your data loader. How can I proceed?

I'm just doing some tests because tomorrow I'm starting to record a new custom dataset with a Baxter robot and I want to be sure how to do things later.

Another question: I have a workstation with an A6000 with 48GB. Should I be able to fine-tune with LORA?

kpertsch, 2024-10-08:
Thanks for your question and sorry for the delayed reply!
To register a dataset, you need to create a config that explains to the data loader which keys from your RLDS dataset it should load, e.g.:
```
"your_new_dataset": {
        "image_obs_keys": {"primary": "image", "secondary": None, "wrist": "wrist_image"},
        "depth_obs_keys": {"primary": None, "secondary": None, "wrist": None},
        "state_obs_keys": ["state"],
        "state_encoding": StateEncoding.POS_QUAT,   # not sure whether this is true for your data, but we don't use state anyways
        "action_encoding": ActionEncoding.EEF_POS,   # this probably also doesn't work for your setup since your actions seem 10-dim, but can just define something new
    }
```

Once you added a config like this, you need to register a transform function, which gives you a chance to perform any dataset specific pre-processing on the fly in the data loader. In the most likely case that you don't need to do any additional processing, you can just define a no-op function like for example [here](https://github.com/openvla/openvla/blob/46b752f477cc5773cc1234b2e82c0e2130e4e890/prismatic/vla/datasets/rlds/oxe/transforms.py#L556). Then you just register your new transform with the name of your dataset at the bottom of the file [here](https://github.com/openvla/openvla/blob/46b752f477cc5773cc1234b2e82c0e2130e4e890/prismatic/vla/datasets/rlds/oxe/transforms.py#L845).

Now you should be all set and you can load your dataset simply by specifying its dataset name in the config here: https://github.com/openvla/openvla/blob/46b752f477cc5773cc1234b2e82c0e2130e4e890/vla-scripts/finetune.py#L82

kpertsch, 2024-10-08:
Sorry forgot the second question: yes, an A6000 should be sufficient for LoRA finetuning

dr-smgad, 2024-11-12:
@kpertsch, thanks for your clarification above. Regarding adopting a different action space like discrete keyboard inputs (e.g., tfds.features.ClassLabel type in action config for the dataset), could you explain more about the steps needed to do Open-VLA fine-tuning on such new action space?

billamiable, 2025-03-12:
> "For option (a), after you converted your dataset to RLDS, you need to register it with our data loader, by registering a dataset config [...]

Hi @ClousTom @kpertsch , I wonder what is the script used to get four files?
- dataset_info.json
- features.json
- example_dataset-train.tfrecord-00000-of-00001
- example_dataset-val.tfrecord-00000-of-00001

billamiable, 2025-03-17:
No worry, I think I figured it out.

mxp1234, 2025-04-21:
> No worry, I think I figured it out.不用担心，我想我已经明白了。

how do you solve it？I have no idea