openvla/openvla #121: How to transform custom dataset into RLDS format with multiprocessing code?

LPY1219, 2024-09-24:
Thank you for your excellent work!

I am currently working on deploying OpenVLA using the CALVIN simulation benchmark. As part of this, I need to convert the CALVIN dataset into the RLDS format. Given the dataset’s size, I need to use a multiprocessing approach for the transformation. However, when attempting to use the official multiprocessing code [here](https://github.com/kpertsch/rlds_dataset_builder/blob/main/example_dataset/example_dataset_dataset_builder.py)  I encounter the following error:
`RuntimeError: Unable to pickle fn CallableWrapperDoFn(<function Map.<locals>.<lambda> at 0x7fe5ac358ca0>): Queue objects should only be shared between processes through inheritance`.

The issue seems to originate from the multiprocessing code:
```
         beam = tfds.core.lazy_imports.apache_beam
         return (
                 beam.Create(episode_paths)
                 | beam.Map(_parse_example)
        )
```
The` _parse_example` function, which uses an `env` object (created with` env = get_env(path, show_gui=False)`), needs to be pickled, but the `env` object cannot be pickled.

Could you please advise on how to perform multiprocessing for this transformation when the parsing function contains non-pickleable objects?
.

LPY1219, 2024-09-29:
I find using the multithreaded branch can solve the problem.

closed 2024-09-29 by LPY1219

zwlanpishu, 2024-11-30:
@LPY1219 Could you please share the Calvin dataset after your conversion? Thank you very much.