openvla/openvla #127: Can I set pretrained=False for vit_large_patch14_reg4_dinov2.lvd142m and ViT-SO400M-14-SigLIP if I want to train openvla with trained openvla-7b-prismatic？

Icecream-blue-sky, 2024-09-27:
When I want to train OpenVLA pretrained with openvla-7b-prismatic, it always raise network error  for downloading model(vit_large_patch14_reg4_dinov2.lvd142m and ViT-SO400M-14-SigLIP)  from hugging face. It is caused by following code:
[image]
Since the vit_large_patch14_reg4_dinov2.lvd142m and ViT-SO400M-14-SigLIP already exist in pretrained openvla-7b-prismatic model, I wonder if is neccesary to init weights for vit_large_patch14_reg4_dinov2.lvd142m and ViT-SO400M-14-SigLIP (set pretrained=True) ?
And when I try to forcibly load the model using the downloaded weights (such as vit_large_patch14_reg4_dinov2.lvd142m/model.safetensors), it shows a size mismatch error.
```python
self.dino_featurizer: VisionTransformer = timm.create_model(                                                                                                                               self.dino_timm_path_or_url, pretrained=False, num_classes=0, img_size=self.default_image_size, checkpoint_pa
th="xxx/vit_large_patch14_reg4_dinov2.lvd142m/model.safetensors"
```
[image]
It's really confused...

moojink, 2024-10-09:
Hi @Icecream-blue-sky,

Unfortunately, it's not clear why those `timm.create_model()` calls are erroring for you (it seems to work fine on my end). And this issue is specific to the timm library, so you may need to post an issue in the pytorch-image-models repo to get support.

If you are still stuck, you can try pasting some code to reproduce the issue and we could try running it on our end to see if it works.

-Moo Jin

closed 2024-10-09 by kpertsch

vasfgg, 2024-11-18:
@Icecream-blue-sky Hello, I met the same problem, how did you solve it?

Icecream-blue-sky, 2024-11-18:
> @Icecream-blue-sky Hello, I met the same problem, how did you solve it?
You can try this.
```python
# Initialize both Featurizers (ViTs) by downloading from HF / TIMM Hub if necessary
        self.dino_featurizer: VisionTransformer = timm.create_model(
            self.dino_timm_path_or_url, pretrained=True, num_classes=0, img_size=self.default_image_size,
            pretrained_cfg={'file':"/share_space/checkpoint/hub/models--timm--vit_large_patch14_reg4_dinov2.lvd142m/snapshots/2718d189d269d0bf6a0d62ed7aa920cdc688ec84/model.safetensors"}
        )
        self.dino_featurizer.eval()

        self.siglip_featurizer: VisionTransformer = timm.create_model(
            self.siglip_timm_path_or_url, pretrained=True, num_classes=0, img_size=self.default_image_size,
            pretrained_cfg={'file':'/share_space/checkpoint/hub/models--timm--ViT-SO400M-14-SigLIP/snapshots/9179d15177ece40964c50492136eda2f3e0c9f61/open_clip_model.safetensors'}
        )
        self.siglip_featurizer.eval()
```