openvla/openvla #129: Finetuning code on LIBERO

4078730, 2024-10-01:
Thank you for your excellent work!

In the "OpenVLA Fine-Tuning Results" section, results for "Octo fine-tuned" are provided. Is there any code, configuration, and rollout scripts available for fine-tuning Octo with LIBERO data?

Thank you for your time and assistance.

TousakaNagio, 2024-10-08:
I guess you only need to modify the path to dataset they have converted in the LoRA fine-tuning script. It works for me.

TousakaNagio, 2024-10-08:
But I am also curious about the fine-tuning configuration to reproduce the results on LIBERO.

kpertsch, 2024-10-09:
For Octo hyperparameters, I just used the official Octo finetuning script, running it on the LIBERO dataset (Octo and OpenVLA use the same data loader, so you don't need to change the format)

closed 2024-10-09 by kpertsch

4078730, 2024-10-10:
Thank you all for your helpful responses and detailed insights! 
I’ll try the same as well.