openvla/openvla #135: We have some questions we'd like to ask you.

mxhcuzz, 2024-10-12:
Hello, after reviewing your OpenVLA project, we are very interested and have a few questions to ask.

Our task is to pick up a yellow object and place it in a blue box.

Image 1 shows the overall environment, and Image 2 is the image captured by the D435 camera, which is sent to the VLA.

We built our environment and robotic arm, but the test results using the initial weights you provided were poor.
We collected around 120 samples and fine-tuned the model, but the test results were still poor, and it seems the robotic arm doesn’t know where the target is.
The robotic arm uses a right-hand coordinate system, and for each task, we collect 300 steps at a frequency of 50Hz, totaling 6 seconds of data.
Do you have any insights into why the results are poor? Could you help us analyze and clarify this situation?
<img width="627" alt="24dda67d083581d20a6144c5df03cf8" src="https://github.com/user-attachments/assets/b91b010d-4363-4051-91ea-7967dafcc51d"
[image]
[image]

kpertsch, 2024-10-15:
Thanks for trying OpenVLA and sorry to hear that it didn't work right away.
A few comments:
- it is expected that the model does not work out of the box -- it's not conditioned on any robot information, so it requires finetuning with your robot data to work well
- your setup looks fine and the task easy enough, I would expect that OpenVLA is able to learn the task (with reasonable data coverage)
- two basic sanity checks: 
 (1) can you replay the actions from one of your collected demonstrations on the robot to make sure it can still perform the task (to make sure your data collection setup is correct)
 (2) can you run images from your finetuning dataset through your inference pipeline (as if they were coming from the robot) and verify that you can match the MSE / token accuracy you see from the model during finetuning (to make sure there are no bugs in your inference pipeline)

- if both tests pass, one other thing I would suggest is to try downsampling your actions to a lower frequency. since OpenVLA does not use action chunking, it may struggle with high-frequency control. Try downsampling the actions in your dataset by 10x (ie 5Hz control), verify that you can still replay demonstrations successfully at that frequency and then finetune OpenVLA with that. This is closer to the control frequency we tested in our paper

Hope this helps!

closed 2024-10-15 by kpertsch

kpertsch, 2024-10-15:
Now also added a performance troubleshooting section the README: https://github.com/openvla/openvla/blob/main/README.md#vla-performance-troubleshooting

qingquluofeng, 2024-10-18:
@mxhcuzz Hello，I met the same problem. Can I ask you some questions?

mxhcuzz, 2024-10-23:
> @mxhcuzz Hello，I met the same problem. Can I ask you some questions?

sure，but I haven't solved it yet

siyuhsu, 2025-01-17:
@mxhcuzz Hi, any updates on finetuning your robot? Have you happened to succeed? Do you have experience to share?