openvla/openvla #138: The train/test split ratio of LIBERO Benchmark evaluation.

jis3613, 2024-10-14:
Hi ! 
Thanks for opening the code for the LIBERO benchmark evaluation.

I succeeded in fine-tuning and evaluating the model on LIBERO dataset.
I wonder how the train-set and the test-set are split in the LIBERO benchmark.
As far as I know, each task suite of LIBERO has 10 tasks and each task with 50 episodes which results in 500 episodes per each task-suite.
How are these episodes split into the train/test set ?
What is the split ratio of train/test set ?

Thanks in advance!

moojink, 2024-10-17:
Hi @jis3613,

We actually just used the full dataset as the training set (no test set) so that we could ensure fair comparisons with other works. However, if you need a validation/test set for hyperparameter tuning, you could split the dataset during project iteration and then later on train on the full dataset when reporting results (it's really up to you). Note that the train sets include less than 500 demos per suite because we filtered out unsuccessful demos.

-Moo Jin

YEOMJINSEOP, 2024-10-25:
Thanks for the helpful issue and reply.

I would like to ask a follow-up question related to this.

**What I understood so far:**

1. Each task suite in LIBERO (spatial, object, goal, 100) consists of 10 tasks, each with 50 episodes.
2. OpenVLA used the LIBERO dataset, which has been filtered and modified.
3. As Moo Jin explained, OpenVLA used the full dataset as the training set.

**Here is my question**: In Appendix E of the LIBERO paper, it states that _"all methods are evaluated across 500 trials for each task suite."_ Does this mean that the same set of tasks and episodes are used for both training and evaluation (test) in OpenVLA? I want to clarify whether there is any distinction between the training set and the evaluation set in this context.

Thanks in advance for your clarification!

jis3613, 2024-10-28:
I think the same tasks and episodes are used for both training and evaluation.

The difference between training and evaluation is that when training, the ground-truth action label, the action trajectory annotated by the humans, are exploited to calculate the loss and train the model.

However, when evaluating, it does not require any GT action label.
It only gives the initial state of each episode, and just let the policy model decide which action to take on its own at each time-step.

It might seem a little unfair that it uses the same initial states of the episodes used during training at evaluation-time,
but I think this is how things are like in the field of robotics.

I am new to robotics research like you, so hopefully the authors can give better answer than mine...

moojink, 2024-10-28:
Hi @YEOMJINSEOP and @jis3613,

For LIBERO, the initial object/robot positions at test time are different from the ones used at train time. Bo Liu (first author of the LIBERO benchmark) confirmed this in [this GitHub issue](https://github.com/Lifelong-Robot-Learning/LIBERO/issues/2#issuecomment-1596101904): "1 task has 50 demonstrations and 50 initial states, the initial states are not necessarily the same as the initial state in the demonstration, but they are sampled i.i.d from the same initial state distribution. By doing so, we can test the in-distribution generalization." You can check this for yourself by saving an image of the initial states at train time and test time, for all episodes within a task, and comparing them.

If you load in the initial object/robot states seen in the training demonstrations, you will observe very high success rates, as OpenVLA fits the training data very well. The numbers we reported in the paper are lower though because we tested on the default eval settings, where the initial states are different than the ones seen in the training demonstrations.

-Moo Jin

moojink, 2024-10-28:
@jis3613 No worries! This information is not immediately clear unless you have played around with the benchmark for a while. Good to hear!

YEOMJINSEOP, 2024-10-28:
> @jis3613 No worries! This information is not immediately clear unless you have played around with the benchmark for a while. Good to hear!

Hi @moojink and @jis3613,

Thank you both for your detailed explanations. 
Your insights clarified a lot about the OpenVLA's LIBERO benchmark setup and evaluation. 
I really appreciate your precious time and help!

Best,
JINSEOP

closed 2024-12-06 by kpertsch