openvla/openvla #139: Question about fine-tuning on real robot advices

Yikai1, 2024-10-14:
Thanks for your reply and advice in #105!  I did more experiments. I am trying to fine-tune in a real ur3 robot.  Some key informations:

- trained on one task: Move the black box right (randomized the cube position)

- train on 80 epoch for this task

- trained on 4090 RTX with 24GB vRAM

- real robot: ur3

```shell
torchrun --standalone --nnodes 1 --nproc-per-node 1  vla-scripts/finetune.py \
  --vla_path "/home/ubuntu/openvla/openvla-7b" \
  --data_root_dir /home/ubuntu/openvla \
  --dataset_name ur3 \
  --run_root_dir /home/ubuntu/openvla/run/test \
  --adapter_tmp_dir /home/ubuntu/openvla/adapter-tmp \
  --lora_rank 32 \
  --batch_size 2 \
  --grad_accumulation_steps 16 \
  --learning_rate 5e-4 \
  --image_aug True \
  --wandb_project openvla \
  --save_steps 300 \
  --max_steps 6000
``` 
Train logs look like that:
[image]
[image]
[image]
Here is a video of the fine-tune dataset (the robot will come to the left of the black box and move it to the right):

https://github.com/user-attachments/assets/ca42e125-c6e1-4eec-aa73-7f66d52e8e13

During the inference (sorry for forgetting to record the video and I do check the subtle errors in the inference pipeline as you said), the robot arm will correctly come to the left of the black box, but will not be able to move it to the right. Do you have any advice for improving our fine-tuning setup that may help to successfully complete the task? What number of episodes and steps per episode do you recommend for such task? Do data like this that do not appear in the pre-training data require more epochs to fine-tune?

And I am also trying to modify your fine_tune.py to use fsdp instead of ddp (for runing fine-tune on multi-GPU for a larger dataset). But here is a bug as follows. Have you tried use fsdp in fine-tune?
```
  File "/home/ubuntu/openvla/openvla/vla-scripts/finetune_fsdp.py", line 470, in <module>
    finetune()
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/draccus/argparsing.py", line 203, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/ubuntu/openvla/openvla/vla-scripts/finetune_fsdp.py", line 375, in finetune
    output: CausalLMOutputWithPast = vla(
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/distributed/fsdp/fully_sharded_data_parallel.py", line 849, in forward
    output = self._fsdp_wrapped_module(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/peft/peft_model.py", line 642, in forward
    return self.get_base_model()(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/distributed/fsdp/fully_sharded_data_parallel.py", line 849, in forward
    output = self._fsdp_wrapped_module(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ubuntu/.cache/huggingface/modules/transformers_modules/openvla-7b/modeling_prismatic.py", line 366, in forward
    patch_features = self.vision_backbone(pixel_values)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/distributed/fsdp/fully_sharded_data_parallel.py", line 849, in forward
    output = self._fsdp_wrapped_module(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ubuntu/.cache/huggingface/modules/transformers_modules/openvla-7b/modeling_prismatic.py", line 121, in forward
    patches, patches_fused = self.featurizer(img), self.fused_featurizer(img_fused)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/distributed/fsdp/fully_sharded_data_parallel.py", line 849, in forward
    output = self._fsdp_wrapped_module(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/ubuntu/.cache/huggingface/modules/transformers_modules/openvla-7b/modeling_prismatic.py", line 43, in wrapper
    result = fn(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/timm/models/vision_transformer.py", line 644, in get_intermediate_layers
    outputs = self._intermediate_layers(x, n)
  File "/home/ubuntu/anaconda3/envs/openvla-ds-kyk/lib/python3.10/site-packages/timm/models/vision_transformer.py", line 617, in _intermediate_layers
    outputs, num_blocks = [], len(self.blocks)
TypeError: object of type 'FullyShardedDataParallel' has no len()
```

kpertsch, 2024-10-15:
Your task seems fine -- can you take a look at my answer here: https://github.com/openvla/openvla/issues/135
If you have already completed both sanity checks, could the problem maybe be in the control frequency of your robot? Control frequencies around 5Hz may work best since OpenVLA is not trained with action chunking.
Also make sure that your demos don't contain "breaks" / very slow actions since the lack of action chunking may lead to OpenVLA getting stuck in states which often have very small actions in the training data.

closed 2024-10-15 by kpertsch

kpertsch, 2024-10-15:
Now also added a section about performance troubleshooting to the readme: https://github.com/openvla/openvla/blob/main/README.md#vla-performance-troubleshooting

Yikai1, 2024-10-16:
> Your task seems fine -- can you take a look at my answer here: #135 If you have already completed both sanity checks, could the problem maybe be in the control frequency of your robot? Control [...]

Thank you very much for the answer. I will continue work on OpenVLA.

leonwu117, 2025-08-10:
Hello,  @Yikai1 . Have you finally solved the problem of deploying OpenVLA on UR3? Can you add WeChat to communicate with me? I am also a graduate student at Shenzhen University.

Yikai1, 2025-08-15:
> e

Can you send me a message through the email on my profile?

Yikai1, 2025-08-15:
> Hello, [@Yikai1](https://github.com/Yikai1) . Have you finally solved the problem of deploying OpenVLA on UR3? Can you add WeChat to communicate with me? I am also a graduate student at Shenzhen University.

Can you send me a message through the email on my profile?