openvla/openvla #141: The regenerated LIBERO dataset has a different dataset size compared to the provided dataset

AndyCao1125, 2024-10-15:
Dear authors,

I carefully followed the regenerated process provided in [rlds_conversion](https://github.com/moojink/rlds_dataset_builder), thanks for your guidance! However, I found that the regenerated dataset has a different dataset size compared to the provided dataset. 

I think it is possible, for example, that the random seed of the underlying simulation environment causes the trajectory length to change when the data is re-collected. But I still wanna check with the authors, is it possible to regenerate the data set with a slight difference in length? If possible, what exactly caused this phenomenon? 

Thank you very much!

moojink, 2024-10-17:
Hi @AndyCao1125,

How different is your regenerated dataset size compared to the one we provided? We are simply replaying the original demonstrations (not changing the lengths of trajectories), so the total number of steps in each dataset should be fixed -- unless for some reason a demonstration that succeeded in our regeneration failed in yours (or vice versa).

We already fix the environment seed (as shown [here](https://github.com/openvla/openvla/blob/0214a0c7c09942fb8e0ec3c3948c00e4e8949911/experiments/robot/libero/libero_utils.py#L24)), since we did notice during our development that the seed matters.

-Moo Jin

AndyCao1125, 2024-10-23:
@moojink 

Thanks for your reply. The regenerated LIBERO Goal dataset has 51584 `num_transitions` with 425 `num_trajectories`, and the provided dataset has 52042 `num_transitions` with 428 `num_trajectories`. I think the difference between the two is very slight, and it may be that some demos fail in my case. However, upon evaluation, these newly generated data sets can also maintain similar results, so this error may be ignored. Thanks!

moojink, 2024-10-28:
@AndyCao1125 Good to know! Thank you for the update.

closed 2024-10-28 by moojink