openvla/openvla #142: LIBERO Reproducibility

jis3613, 2024-10-17:
Hi, it's me again!

I tried to reproduce the LIBERO benchmark results, but failed.

|Model|Spatial|Object|Goal|Long|Average|
|------|------|------|------|------|------|
|Reported (3trials mean)|84.7|88.4|79.2|53.7|76.5|
|Reproduced (single trial)|83|88.2|74.2|48.4|73.75|

Considering our result is only single-trial, still our results for goal and long are bit out of range from the reported results.

I followed the training script you provided for training on Bridge dataset.
This is the script I ran.
`
torchrun --standalone --nnodes 1 --nproc-per-node 1 vla-scripts/finetune.py \
            --vla_path openvla/openvla-7b \
            --data_root_dir datasets \
            --dataset_name libero_10_no_noops \
            --run_root_dir runs \
            --adapter_tmp_dir adapter_tmp_weights \
            --lora_rank 32 \
            --batch_size 4 \
            --grad_accumulation_steps 4 \
            --learning_rate 5e-4 \
            --image_aug True \
            --save_steps 5000 \
`
We are traininig on a single A100 GPU.

Do you use different hyper-parameter from this script?
If it is, it  would be really helpful if you can provide the information (e.g. learning rate, batch size, .etc).

Thanks!

moojink, 2024-10-17:
Hi @jis3613,

We listed the hyperparameters in [this issue](https://github.com/openvla/openvla/issues/112#issuecomment-2401300268) (you can also find the same details on the Hugging Face model cards, e.g., [this one](https://huggingface.co/openvla/openvla-7b-finetuned-libero-spatial) for LIBERO-Spatial). Note that we used 8 A100 GPUs with 80GB VRAM each so that the training jobs finish quicker, but it's not absolutely necessary.

Hope that helps you reproduce the results!

-Moo Jin

moojink, 2024-10-17:
@jis3613 Another thing worth trying is to download our fine-tuned checkpoints and see if you can reproduce the numbers that we reported (as a sanity check that there is no issue in the test environment, as it's possible that environment-rendered images could look different on different hardware, which would lead to slight differences in performance).

jis3613, 2024-10-17:
Thank you so much Moojin! Hope to see you in person in near future at a conference!

jis3613, 2024-10-18:
Hi moojin! I have a one quick question.

If I run the following training scrip, would the total batch size be 64 ?
64 = 4 (batch_size) * 4 (grad_accumulation_steps) * 4 (nproc-per-node)

gpu_id=0,1,2,3
CUDA_VISIBLE_DEVICES=${gpu_id} torchrun --standalone --nnodes 1 --nproc-per-node 4 vla-scripts/finetune.py \
            --vla_path openvla/openvla-7b \
            --data_root_dir datasets \
            --dataset_name libero_object_no_noops \
            --run_root_dir runs \
            --adapter_tmp_dir adapter_tmp_weights \
            --lora_rank 32 \
            --batch_size 4 \
            --grad_accumulation_steps 4 \
            --learning_rate 5e-4 \
            --image_aug True \
            --save_steps 5000 \

moojink, 2024-10-28:
@jis3613 Yes, that is correct!

You can play around with batch size depending on how many GPUS / how much VRAM is available to you. Sometimes a small batch size could still work (so it is worth trying `grad_accumulation_steps == 1` to see if training is stable enough without gradient accumulation). Gradient accumulation will slow down each gradient step, so I would only use it if training loss oscillates too much without it.

Robert-hua, 2024-12-02:
I attempted to reproduce the results of the LIBERO benchmark following the provided README instructions. However, the reproduced results appear to deviate from the reported results. Below are the details of the comparison:
| Model                | Spatial | Object | Goal | Long | Average |
|----------------------|---------|--------|------|------|---------|
| **Reported (3 trials mean)** | 84.7    | 88.4   | 79.2 | 53.7 | 76.5    |
| **Reproduced (single trial)** | 83.4    | 84.4   | 78.0 | 51.4 | 74.3    |

To reproduce the benchmark, I used the four Hugging Face open-source models specified in the README to run the corresponding tasks. Considering that I conducted only a single-trial, I wonder if the lack of three-trials might account for the discrepancy between my results and the reported ones.

Could there be other factors or details I might have overlooked that contributed to the difference? Any guidance on this would be greatly appreciated.

moojink, 2024-12-03:
Hi @Robert-hua,

We observed little variance in the success rates between the 3 trials for OpenVLA policies when we tested them (the variance was larger for other methods, like Octo and Diffusion Policy). Therefore, I would guess that the discrepancy is caused by something else.

Are you using Python 3.10.14 and Torch 2.2.0? These are the versions we tested with.

Also, which GPU are you running the evaluations with? The LIBERO success rates we reported were obtained by running the provided OpenVLA checkpoints on a machine with an A100 GPU. If you are using a different GPU, it's possible that the resulting numbers are slightly different due to potential differences in the implementations of floating-point operations.

-Moo Jin

Robert-hua, 2024-12-03:
Thank you for your response. My testing environment is set up with Python 3.10.15 and Torch 2.2.0, which I believe should have minimal impact on the test results. However, I am using an RTX 3090 GPU, which might influence the outcomes. Should I switch to an A100 for testing instead?

moojink, 2024-12-09:
Hi @Robert-hua,

From our experience, we have seen the best results when testing checkpoints using the same GPU that we used for training. So if you are testing your own checkpoints, I would recommend that you to use the GPU(s) you trained them with (e.g., train with RTX 3090 -> test with RTX 3090 for best results). If you are evaluating the LIBERO checkpoints we provided and having trouble reproducing the numbers that we reported in the paper, then yes, I would recommend trying to deploy the checkpoints on an A100 GPU to see if that helps.

-Moo Jin

SoshinK, 2024-12-23:
Hi @moojink! Thank you for excellent work! Could you also provide hyperparameters for Octo?

Thank You!

closed 2025-02-03 by kpertsch

BarneetWei, 2025-08-20:
> [@jis3613](https://github.com/jis3613) Another thing worth trying is to download our fine-tuned checkpoints and see if you can reproduce the numbers that we reported (as a sanity check that there is [...]

 Not reproduce the numbers that you reported.
I used H20 to run the experiments on LIBERO 
Wed Aug 20 10:58:37 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 555.42.06              Driver Version: 555.42.06      CUDA Version: 12.5     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA H20                     Off |   00000000:F1:09.0 Off |                    0 |
| N/A   49C    P0            315W /  500W |   72733MiB /  97871MiB |    100%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA H20                     Off |   00000000:F3:09.0 Off |                    0 |
| N/A   40C    P0            281W /  500W |   56194MiB /  97871MiB |    100%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+
|   2  NVIDIA H20                     Off |   00000000:F5:09.0 Off |                    0 |
| N/A   47C    P0            289W /  500W |   56256MiB /  97871MiB |     98%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+
|   3  NVIDIA H20                     Off |   00000000:F7:09.0 Off |                    0 |
| N/A   39C    P0            280W /  500W |   55120MiB /  97871MiB |     70%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+
|   4  NVIDIA H20                     Off |   00000000:F9:09.0 Off |                    0 |
| N/A   48C    P0            289W /  500W |   72841MiB /  97871MiB |    100%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+
|   5  NVIDIA H20                     Off |   00000000:FB:09.0 Off |                   49 |
| N/A   41C    P0            287W /  500W |   56282MiB /  97871MiB |     67%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+
|   6  NVIDIA H20                     Off |   00000000:FD:09.0 Off |                    0 |
| N/A   45C    P0            279W /  500W |   56192MiB /  97871MiB |    100%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+
|   7  NVIDIA H20                     Off |   00000000:FF:09.0 Off |                    0 |
| N/A   39C    P0            267W /  500W |   54948MiB /  97871MiB |     69%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+