openvla/openvla #144: JOINT_POS not compatible

ClousTom, 2024-10-24:
Hello. I have registered my dataset in this way:

"baxter_dflacon": {  # My dataset from Baxter (grasp the detergent flacon)
    "image_obs_keys": {"primary": "image", "secondary": "head_image", "wrist": None},
    "depth_obs_keys": {"primary": None, "secondary": None, "wrist": None},
    "state_obs_keys": ["state"],
    "state_encoding": StateEncoding.JOINT,
    "action_encoding": ActionEncoding.JOINT_POS,
}
as my representation is:

joint angles (7) + gripper state (1), for state
joint delta position (7) + gripper state (1), for action
but running the finetuning command I receive this error:
10/21 [19:06:34] WARNING | >> [*] Skipping baxter_dflacon materialize.py:129
due to Error: Cannot load
baxter_dflacon; only EEF_POS &
EEF_R6 actions supported!

why? is "action_encoding": ActionEncoding.JOINT_POS not supported?

ClousTom, 2024-11-13:
Hi. Thank you for your reply.

However, the .JOINT_POS format is among the possible representations described in the ‘configs.py’ file. In addition, there are several datasets in the latter file, which rappresentation is in this format. Example:

`"berkeley_rpt_converted_externally_to_rlds": {
        "image_obs_keys": {"primary": None, "secondary": None, "wrist": "hand_image"},
        "depth_obs_keys": {"primary": None, "secondary": None, "wrist": None},
        "state_obs_keys": ["joint_pos", "gripper"],
        "state_encoding": StateEncoding.JOINT,
        "action_encoding": ActionEncoding.JOINT_POS,
   }`

Like this one, there are many others. Were all datasets in this format used for training? Or were they ignored?

I just did a finetuning with a dataset of about 100 episodes, recorded for a baxter robot, in .JOINT_POS format. The result was very bad. The robotic arm does not go absolutely where it should, but goes upwards and curls in on itself.

In addition, I would like to ask how it is possible to implement the model on two robotic arms, as the .JOINT_POS format gives very bad results, and the only way mentioned to use two joints at the same time is to use a double representation with delta joint angles.

JiayiCheung, 2024-11-13:
> 你好。谢谢你的回复。
> 
> 但是，.JOINT_POS 格式是 'configs.py' 文件中描述的可能表示形式之一。此外，后一个文件中还有几个数据集，其表示形式就是这种格式。例如：
> 
> `"berkeley_rpt_converted_externally_to_rlds": { "image_obs_keys": {"primary": None, "secondary": None, "wrist": "hand_image"}, "depth_obs_keys": {"primary": None, "secondary": None, "wrist": None}, "state_obs_keys": ["joint_pos", "gripper"], "state_encoding": StateEncoding.JOINT, "action_encoding": ActionEncoding.JOINT_POS, }`
> 
> 像这个一样，还有很多其他的。这种格式的所有数据集都用于训练吗？还是被忽略了？
> 
> 我刚刚用一个约 100 个情节的数据集进行了微调，这些情节是为 Baxter 机器人录制的，格式为 .JOINT_POS。结果非常糟糕。机械臂并没有完全移动到它应该去的地方，而是向上弯曲并自行卷起。
> 
> 另外，我想问一下如何在两个机械臂上实现该模型，因为.JOINT_POS 格式会产生非常糟糕的结果，而提到的同时使用两个关节的唯一方法是使用具有增量关节角度的双重表示。

Friend ，Did you meet a problem about SkipDecoding？

moojink, 2024-12-03:
Hi @ClousTom,

We did not use JOINT_POS in the OpenX pretraining phase for OpenVLA. However, the model can still be fine-tuned with action spaces like absolute/relative joint angles. I have personally fine-tuned the OpenVLA model to complete tasks successfully on the [ALOHA robot](https://tonyzhaozh.github.io/aloha/), which has two arms and uses absolute joint angle targets as the action space.

We may release a followup report on how to do this fine-tuning effectively by the end of this year or early next year. Stay tuned!

-Moo Jin

closed 2024-12-06 by kpertsch