openvla/openvla #145: int4 and int8 quantization

ClousTom, 2024-10-28:
From the way it is written in the paper, int4 and int8 quantizations are supported. But how do I set them?
Reading on another [issue](https://github.com/openvla/openvla/issues/10), I should set the code like this:
vla = AutoModelForVision2Seq.from_pretrained(
    “openvla/openvla-7b”, 
    attn_implementation=“flash_attention_2”, # [Optional] Requires `flash_attn`
    low_cpu_mem_usage=True, 
    trust_remote_code=True,
    load_in_4bit=True,
).to(“cuda:0”)

but I get this error:
Loading checkpoint fragments: 100%|████████| 4/4 [00:03<00:00, 1.05it/s].
You should not move a model that is deployed using acceleration hooks.
Traceback (last call):
  File “/home/engineeringroom/Claudio/move_baxter.py”, line 35, in <module>
    ).to(“cuda:0”)
  File “/home/engineeringroom/anaconda3/envs/openvla/lib/python3.10/site-packages/accelerate/big_modeling.py”, line 457, in wrapper
    return fn(*args, **kwargs)
  File “/home/engineeringroom/anaconda3/envs/openvla/lib/python3.10/site-packages/transformers/modeling_utils.py”, line 2670, in to
    raise ValueError(
ValueError: `.to` is not supported for `4-bit` or `8-bit` bitsandbytes templates. Please use the model as is, since the model has already been set to the correct devices and cast in the correct `dtype`.

This is my code:
processor = AutoProcessor.from_pretrained("openvla/openvla-7b", trust_remote_code=True)
vla = AutoModelForVision2Seq.from_pretrained(
    "openVLA_test/checkpoints/openvla-7b+baxter_dflacon+b10+lr-0.0005+lora-r32+dropout-0.0",
    attn_implementation="flash_attention_2",  
    #torch_dtype=torch.bfloat16, 
    low_cpu_mem_usage=True, 
    trust_remote_code=True,
    load_in_4bit=True
).to("cuda:0")

image: Image.Image = get_from_camera()
prompt = "In: What action should the robot take to {<INSTRUCTION>}?\nOut:"

#inputs = processor(prompt, image).to("cuda:0", dtype=torch.bfloat16)
inputs = processor(prompt, image).to("cuda:0")

action = vla.predict_action(**inputs, unnorm_key="baxter_dflacon", do_sample=False)

please, explain me how to set it. I want to run the inference on a 8GB GPU, so I need int4.

khw11044, 2024-11-19:
I want to know about this, too

khw11044, 2024-11-19:
check verify_openvla.py file

```
    # === 8-BIT QUANTIZATION MODE (`pip install bitsandbytes`) :: [~9GB of VRAM Passive || 10GB of VRAM Active] ===
    # print("[*] Loading in 8-Bit Quantization Mode")
    # vla = AutoModelForVision2Seq.from_pretrained(
    #     MODEL_PATH,
    #     attn_implementation="flash_attention_2",
    #     torch_dtype=torch.float16,
    #     quantization_config=BitsAndBytesConfig(load_in_8bit=True),
    #     low_cpu_mem_usage=True,
    #     trust_remote_code=True,
    # )

    # === 4-BIT QUANTIZATION MODE (`pip install bitsandbytes`) :: [~6GB of VRAM Passive || 7GB of VRAM Active] ===
    # print("[*] Loading in 4-Bit Quantization Mode")
    # vla = AutoModelForVision2Seq.from_pretrained(
    #     MODEL_PATH,
    #     attn_implementation="flash_attention_2",
    #     torch_dtype=torch.float16,
    #     quantization_config=BitsAndBytesConfig(load_in_4bit=True),
    #     low_cpu_mem_usage=True,
    #     trust_remote_code=True,
    # )
```

ClousTom, 2024-11-21:
Thank you so much @khw11044 

By the way, without ".to("cuda:0)" at the end, part of the model runs on CPU+RAM. Is it right?

Furthermore, attn_implementation="flash_attention_2" gives me an error like "this option may cause undesired behaviour and reduce performance". Indeed, the robot doesn't move. However, if I remove it, there are no more errors and the robot starts to move.

kpertsch, 2024-12-06:
Hey folks, sorry for the late reply.
It seems that bitsandbytes indeed places the model automatically on GPU already if it's initialized with `load_in_4bit=True`, so you're correct in removing the .to() statement.

closed 2024-12-06 by kpertsch

ganganngannn, 2025-03-10:
```python 
        self.vla = AutoModelForVision2Seq.from_pretrained(
            self.openvla_path,
            attn_implementation=attn_implementation,
            torch_dtype=torch.float16,
            quantization_config=BitsAndBytesConfig(load_in_4bit=True),
            low_cpu_mem_usage=True,
            trust_remote_code=True,
        )
```
my code is same as you，but it`s still the bug:

```
ValueError: `.to` is not supported for `4-bit` or `8-bit` bitsandbytes models. Please use the model as it is, since the model has already been set to the correct devices and casted to the correct `dtype`.

``` 

why?

Om-Kulkarni, 2025-04-08:
I have been trying to run the updated code with int8 quantization but have had no luck. It's the same ValueError as before. Was anyone able to fix this issue?

```
Traceback (most recent call last):
  File "/home/snowwhite/OmKulkarni/OpenVLA/openvla-oft/quick_start.py", line 22, in <module>
    vla = get_vla(cfg)
  File "/home/snowwhite/OmKulkarni/OpenVLA/openvla-oft/experiments/robot/openvla_utils.py", line 282, in get_vla
    vla = AutoModelForVision2Seq.from_pretrained(
  File "/home/snowwhite/miniconda3/envs/openvla-oft/lib/python3.10/site-packages/transformers/models/auto/auto_factory.py", line 558, in from_pretrained
    return model_class.from_pretrained(
  File "/home/snowwhite/miniconda3/envs/openvla-oft/lib/python3.10/site-packages/transformers/modeling_utils.py", line 3735, in from_pretrained
    dispatch_model(model, **device_map_kwargs)
  File "/home/snowwhite/miniconda3/envs/openvla-oft/lib/python3.10/site-packages/accelerate/big_modeling.py", line 501, in dispatch_model
    model.to(device)
  File "/home/snowwhite/miniconda3/envs/openvla-oft/lib/python3.10/site-packages/transformers/modeling_utils.py", line 2670, in to
    raise ValueError(
ValueError: `.to` is not supported for `4-bit` or `8-bit` bitsandbytes models. Please use the model as it is, since the model has already been set to the correct devices and casted to the correct `dtype`.
```

MexWayne, 2025-05-16:
hi, I have made this.
first, you should install:
    (1) pip install bitsandbytes
    (2) pip install transformers --upgrade

then, restart the conda env your made to makesure the package you install is now work on
and you shoudl make sure the infer code and input code as the pic shown

[image]

[image]

third, run the code like as the following pic shown(for me, i got a 4070ti in my pc):
python vla-scripts/extern/verify_openvla.py
and you will get the the action with the random input images

[image]

Massive respect to the openvla, what a great work! and hope my note can help.