openvla/openvla #147: About "output.logits"

MysticFace-0, 2024-10-31:
Dear OpenVLA Authors,

Thank you for your outstanding work.

I have a question regarding output.logits.

I noticed that only a small portion of output.logits at the end is used for supervision: 
output.logits[:, vla.module.vision_backbone.featurizer.patch_embed.num_patches: -1].

Would you please clarify if the earlier parts of the output.logits sequence are indeed not supervised or utilized? Additionally, wouldn't not directly outputting the action logits potentially slow down the action generation process?

Thank you very much for your time and response.

Best regards,

moojink, 2024-12-03:
Hi @MysticFace-0,

Yes, we only consider the 7 robot action tokens when computing the cross-entropy loss that is used for learning. We do not compute the loss for other tokens in the prompt, such as `"What action should the robot take to <do some task>?"` – i.e., we only care about generating the robot action tokens correctly.

> Additionally, wouldn't not directly outputting the action logits potentially slow down the action generation process?

I'm not sure I understand this part. During training, you would get the full `output.logits` anyway since you still have to do the full forward pass given all the inputs. And then we just splice the logits to get the action tokens mask to only focus on the action tokens when computing the loss.

Feel free to follow up if that doesn't make sense!

-Moo Jin

kpertsch, 2024-12-06:
The particular snippet of code you posted in your question just chops off the logits corresponding to image patch *inputs*, which we indeed don't want to apply loss on. As Moo Jin mentioned we are also not applying loss on any of the logits corresponding to prompt tokens, but purely on logits that represent action outputs.

closed 2024-12-06 by kpertsch