openvla/openvla #153: Multi-gpu training of lora_finetune get stucked on "trainable params: xxxxx"

Icecream-blue-sky, 2024-11-11:
Single-gpu training is quite normal. When I try to train with 3 gpu， it gets stucked at "Loading checkpioing shareds 100%. trainable params: 222xx || all params....". Why？
[image]
My command is: torchrun --standalone --nnodes 1 --nproc-per-node 3 vla-scripts/finetune.py --vla_path xxx --data_root_dir xxx --dataset_name fractal20220817_data --lora_rank 64 --batch_size 16 --grad_accumulation_steps 1 --learning_rate 5e-4 --image_aug True --wandb_project openvla_lora_icecream --wandb_entity xxx

kpertsch, 2024-12-06:
It's a bit hard to debug where the code gets stuck -- you could use py-spy to check in what command the code is hanging on your machine.

kpertsch, 2024-12-06:
Will close this issue for now since it's been a while since you asked this question, but if it's still an issue, feel free to re-open!

closed 2024-12-06 by kpertsch

billamiable, 2025-03-24:
try `export NCCL_P2P_LEVEL=NVL`