openvla/openvla #158: LIBERO_modified_RLDS problem

WendyeeWang, 2024-11-20:
(openvla) wangyi@amax:/data/wy/znz/openvla$ torchrun --standalone --nnodes 1 --nproc-per-node 1 vla-scripts/finetune.py --vla_path "openvla/openvla-7b" --data_root_dir "/data/wy/znz/openvla/modified_libero_rlds/libero_spatial_no_noops/1.0.0" --dataset_name libero_spatial_no_noops --run_root_dir "/data/wy/znz/openvla/spatial-self-fintune" --adapter_tmp_dir "/data/wy/znz/openvla/spatial-self-fintune-weight"  --lora_rank 32 --batch_size 8  --grad_accumulati
on_steps 1 --learning_rate 5e-4 --image_aug False --save_steps  2000
2024-11-20 10:31:40.940499: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2024-11-20 10:31:40.988015: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2024-11-20 10:31:40.988059: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2024-11-20 10:31:40.989372: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2024-11-20 10:31:41.001872: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 AVX_VNNI AMX_TILE AMX_INT8 AMX_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-11-20 10:31:41.862828: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
Fine-tuning OpenVLA Model `openvla/openvla-7b` on `libero_spatial_no_noops`
Loading checkpoint shards: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3/3 [00:00<00:00,  5.13it/s]
trainable params: 110,828,288 || all params: 7,652,065,472 || trainable%: 1.4483
Traceback (most recent call last):
  File "/amax/data/wy/znz/openvla/vla-scripts/finetune.py", line 368, in <module>
    finetune()
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/draccus/argparsing.py", line 203, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/amax/data/wy/znz/openvla/vla-scripts/finetune.py", line 215, in finetune
    vla_dataset = RLDSDataset(
  File "/amax/data/wy/znz/openvla/prismatic/vla/datasets/datasets.py", line 140, in __init__
    self.dataset, self.dataset_length, self.dataset_statistics = self.make_dataset(rlds_config)
  File "/amax/data/wy/znz/openvla/prismatic/vla/datasets/datasets.py", line 143, in make_dataset
    return make_interleaved_dataset(**rlds_config)
  File "/amax/data/wy/znz/openvla/prismatic/vla/datasets/rlds/dataset.py", line 510, in make_interleaved_dataset
    _, dataset_statistics = make_dataset_from_rlds(**data_kwargs, train=train)
  File "/amax/data/wy/znz/openvla/prismatic/vla/datasets/rlds/dataset.py", line 202, in make_dataset_from_rlds
    builder = tfds.builder(name, data_dir=data_dir)
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/tensorflow_datasets/core/logging/__init__.py", line 166, in __call__
    return function(*args, **kwargs)
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/tensorflow_datasets/core/load.py", line 215, in builder
    raise not_found_error
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/tensorflow_datasets/core/load.py", line 196, in builder
    cls = builder_cls(str(name))
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/tensorflow_datasets/core/load.py", line 121, in builder_cls
    cls = registered.imported_builder_cls(str(ds_name))
  File "/home/wangyi/miniconda3/envs/openvla/lib/python3.10/site-packages/tensorflow_datasets/core/registered.py", line 301, in imported_builder_cls
    raise DatasetNotFoundError(f'Dataset {name} not found.')
tensorflow_datasets.core.registered.DatasetNotFoundError: Dataset libero_spatial_no_noops not found.
Available datasets:
        - abstract_reasoning
        - accentdb
        - aeslc
        - aflw2k3d
        - ag_news_subset
        - ai2_arc
        - ai2_arc_with_ir
        - amazon_us_reviews
        - anli
        - answer_equivalence
        - arc
        - asqa
        - asset
        - assin2
        - bair_robot_pushing_small
        - bccd
        - beans
        - bee_dataset
        - beir
        - big_patent
        - bigearthnet
        - billsum
        - binarized_mnist
        - binary_alpha_digits
        - ble_wind_field
        - blimp
        - booksum
[... 317 lines cut ...]
        - wmt17_translate
        - wmt18_translate
        - wmt19_translate
        - wmt_t2t_translate
        - wmt_translate
        - wordnet
        - wsc273
        - xnli
        - xquad
        - xsum
        - xtreme_pawsx
        - xtreme_pos
        - xtreme_s
        - xtreme_xnli
        - yahoo_ltrc
        - yelp_polarity_reviews
        - yes_no
        - youtube_vis

Check that:
    - if dataset was added recently, it may only be available
      in `tfds-nightly`
    - the dataset name is spelled correctly
    - dataset class defines all base class abstract methods
    - the module defining the dataset class is imported

closed 2024-11-20 by WendyeeWang

donovanclay, 2024-11-27:
How did you resolve this?

qxiao0521, 2024-12-10:
Have you solved this problem? I've envountered the same problem and haven't found a suitable solution yet. Can you give me some advice?

Zhangwenyao1, 2024-12-11:
pip install -e . in the openvla dir and retry

LukeLIN-web, 2025-02-07:
> pip install -e . in the openvla dir and retry

It still happen.

```
    raise DatasetNotFoundError(f'Dataset {name} not found.')
tensorflow_datasets.core.registered.DatasetNotFoundError: Dataset droid_wipe not found. 

The builder directory /datamodified_libero_rlds/droid_wipe doesn't contain any versions.
No builder could be found in the directory: /data/modified_libero_rlds for the builder: droid_wipe.
No registered data_dirs were found in:
        - /datamodified_libero_rlds
```

LukeLIN-web, 2025-02-07:
I think this is because,      dataset_name: str = "droid_wipe"    default, so we need to specific dataset name . Search globally is helpful.  I search `bridge_orig` I found the demo `torchrun --standalone --nnodes 1 --nproc-per-node 1 vla-scripts/finetune.py \
  --vla_path "openvla/openvla-7b" \
  --data_root_dir <PATH TO BASE DATASETS DIR> \
  --dataset_name bridge_orig \`