openvla/openvla #162: Codebase/Checkpoints for diffusoin policy on libero

YY-GX, 2024-11-24:
Hey, thanks so much for testing openvla on libero data, which is so helpful!! 

I'm now trying to reproduce the results from the table you provided for the libero data. For openvla, it runs well. 

But for diffusion policy, I'm now using diffusion policy in [lerobot](https://github.com/huggingface/lerobot)'s code to test, but I can only reach **0% succ rate** for even a simple task, such as "close the upper drawer'. So I'm wondering what codebase for diffusion policy you're using and any tips/suggestions for training and evaluating diffusion policy on libero data. 

Thanks a lot if you could help!!

moojink, 2024-12-10:
Hi @YY-GX,

For Diffusion Policy, we used the DROID policy learning repo here: https://github.com/droid-dataset/droid_policy_learning

The above is a language-conditioned version of the official Diffusion Policy implementation here: https://github.com/real-stanford/diffusion_policy

We unfortunately do not plan to release code/checkpoints for baseline methods but can provide some details that may be helpful. For Diffusion Policy on LIBERO, here are some details:
* Trained for 25-50 epochs per task suite independently
* Batch size 128
* 2-step observation history, 8-step prediction horizon with 3-step action horizon (i.e., predict 8 actions and execute 3)
* DDIM sampler with 100 diffusion steps
* 224x224 px images (third-person only) + robot proprio state as input
* Image augmentations: Color jitter + random crop (with 202x202 px crops)

-Moo Jin

closed 2024-12-11 by moojink

hqzzzhang, 2024-12-19:
@YY-GX I also met this dilemma, could you solve it ?

hqzzzhang, 2024-12-19:
I also met this dilemma, have you solved it?

hqzzzhang, 2024-12-19:
@YY-GX I also met this dilemma, have you solved it?

g0kul6, 2025-01-23:
> Hi [@YY-GX](https://github.com/YY-GX),
>
> For Diffusion Policy, we used the DROID policy learning repo here: https://github.com/droid-dataset/droid_policy_learning
>
> The above is a language-conditioned [...]

I got the results by following the above steps except for LIBERO Goal. Can you let me know what hyperparams you used for the LIBERO Goal with diffusion policy?

yfpeng1234, 2025-02-11:
Hi @g0kul6 , I'm also trying to reproduce the result of diffusion policy on Libero benchmark.Could you please share which dataset you use for training diffusion policy? The filtered Libero dataset used by the OpenVLA codebase or the original Libero dataset?
Thanks!

g0kul6, 2025-03-27:
> Hi [@g0kul6](https://github.com/g0kul6) , I'm also trying to reproduce the result of diffusion policy on Libero benchmark.Could you please share which dataset you use for training diffusion policy? The filtered Libero dataset used by the OpenVLA codebase or the original Libero dataset? Thanks!

I use the filtered dataset.

Lucky-Light-Sun, 2025-04-01:
> Hi [@YY-GX](https://github.com/YY-GX),
>
> For Diffusion Policy, we used the DROID policy learning repo here: https://github.com/droid-dataset/droid_policy_learning
>
> The above is a language-conditioned [...]

hello, have you experimented with Diffusion Policy using only image inputs (without robot state information)?