openvla/openvla #166: Fine Tuning on new RLDS dataset

artalogica, 2024-11-29:
I added the dataset to the dataset config and the dataset transformation function but I get this error when I try to Lora finetune: 

[2024-11-28 20:15:56,671] torch.distributed.run: [WARNING] 
[2024-11-28 20:15:56,671] torch.distributed.run: [WARNING] *****************************************
[2024-11-28 20:15:56,671] torch.distributed.run: [WARNING] Setting OMP_NUM_THREADS environment variable for each process to be 1 in default, to avoid your system being overloaded, please further tune the variable for optimal performance in your application as needed. 
[2024-11-28 20:15:56,671] torch.distributed.run: [WARNING] *****************************************
2024-11-28 20:16:00.491334: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2024-11-28 20:16:00.491336: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2024-11-28 20:16:00.982189: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2024-11-28 20:16:00.982187: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2024-11-28 20:16:00.982231: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2024-11-28 20:16:00.982254: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2024-11-28 20:16:01.050653: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2024-11-28 20:16:01.050652: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2024-11-28 20:16:01.221068: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-11-28 20:16:01.221063: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2024-11-28 20:16:02.726744: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
2024-11-28 20:16:02.726797: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT

Loading checkpoint shards:   0%|          | 0/3 [00:00<?, ?it/s]
Loading checkpoint shards:   0%|          | 0/3 [00:00<?, ?it/s]
Loading checkpoint shards:  33%|███▎      | 1/3 [00:00<00:01,  1.62it/s]
Loading checkpoint shards:  33%|███▎      | 1/3 [00:00<00:01,  1.60it/s]
Loading checkpoint shards:  67%|██████▋   | 2/3 [00:00<00:00,  2.19it/s]
Loading checkpoint shards:  67%|██████▋   | 2/3 [00:00<00:00,  2.14it/s]
Loading checkpoint shards: 100%|██████████| 3/3 [00:01<00:00,  2.84it/s]
Loading checkpoint shards: 100%|██████████| 3/3 [00:01<00:00,  2.52it/s]

Loading checkpoint shards: 100%|██████████| 3/3 [00:01<00:00,  2.86it/s]
Loading checkpoint shards: 100%|██████████| 3/3 [00:01<00:00,  2.52it/s]
Traceback (most recent call last):
  File "/home/suli/openvla/vla-scripts/finetune.py", line 327, in <module>
Traceback (most recent call last):
  File "/home/suli/openvla/vla-scripts/finetune.py", line 327, in <module>
    finetune()
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/draccus/argparsing.py", line 203, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/suli/openvla/vla-scripts/finetune.py", line 195, in finetune
    vla_dataset = RLDSDataset(
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/prismatic/vla/datasets/datasets.py", line 92, in __init__
    per_dataset_kwargs, weights = get_oxe_dataset_kwargs_and_weights(
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/prismatic/vla/datasets/rlds/oxe/materialize.py", line 116, in get_oxe_dataset_kwargs_and_weights
    make_oxe_dataset_kwargs(
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/prismatic/vla/datasets/rlds/oxe/materialize.py", line 31, in make_oxe_dataset_kwargs
    dataset_kwargs = deepcopy(OXE_DATASET_CONFIGS[dataset_name])
KeyError: 'example_dataset'
    finetune()
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/draccus/argparsing.py", line 203, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/suli/openvla/vla-scripts/finetune.py", line 195, in finetune
    vla_dataset = RLDSDataset(
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/prismatic/vla/datasets/datasets.py", line 92, in __init__
    per_dataset_kwargs, weights = get_oxe_dataset_kwargs_and_weights(
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/prismatic/vla/datasets/rlds/oxe/materialize.py", line 116, in get_oxe_dataset_kwargs_and_weights
    make_oxe_dataset_kwargs(
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/prismatic/vla/datasets/rlds/oxe/materialize.py", line 31, in make_oxe_dataset_kwargs
    dataset_kwargs = deepcopy(OXE_DATASET_CONFIGS[dataset_name])
KeyError: 'example_dataset'
[2024-11-28 20:17:26,797] torch.distributed.elastic.multiprocessing.api: [ERROR] failed (exitcode: 1) local_rank: 0 (pid: 1989981) of binary: /home/suli/miniforge3/envs/finetune/bin/python3.10
Traceback (most recent call last):
  File "/home/suli/miniforge3/envs/finetune/bin/torchrun", line 8, in <module>
    sys.exit(main())
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 347, in wrapper
    return f(*args, **kwargs)
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/torch/distributed/run.py", line 812, in main
    run(args)
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/torch/distributed/run.py", line 803, in run
    elastic_launch(
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 135, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/home/suli/miniforge3/envs/finetune/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 268, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
vla-scripts/finetune.py FAILED
------------------------------------------------------------
Failures:
[1]:
  time      : 2024-11-28_20:17:26
  host      : babel-7-17.ib
  rank      : 1 (local_rank: 1)
  exitcode  : 1 (pid: 1989982)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2024-11-28_20:17:26
  host      : babel-7-17.ib
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 1989981)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
============================================================

this was my addition to OXE_DATASET_CONFIGS: 
 "example_dataset": {
        "image_obs_keys": {"primary": "image", "secondary": None, "wrist": None},
        "depth_obs_keys": {"primary": None, "secondary": None, "wrist": None},
        "state_obs_keys": ["eef_state", None, None],
        "state_encoding": StateEncoding.POS_EULER,
        "action_encoding": ActionEncoding.EEF_POS,

    }, 
    
 and I did not transform my dataset since it fits the training format.

JiayiCheung, 2024-11-30:
I meet this problem too

kpertsch, 2024-12-06:
Can you please verify that you added an entry with the name "example_dataset" to OXE_DATASET_CONFIGS?
Maybe you just created the config, but didn't register it in the dictionary at the bottom of the file?

I will close this issue for now, since it's been a little while since you raised it, but feel free to reopen if it's still an issue!

closed 2024-12-06 by kpertsch