openvla/openvla #167: Memory usage gradually increases during traing

LPY1219, 2024-12-03:
Hi,
Thanks for your great work.  Recently, I’ve been working on reproducing the LIBERO results. However, I’ve encountered an issue where the CPU memory usage (not GPU memory) gradually increases during training. Eventually, it reaches the maximum limit, causing the entire program to crash.

I suspect the problem might be related to dataset loading or memory management. Could you kindly provide some advice or recommendations to address this potential memory leak?

[image]

betray12138, 2024-12-05:
Hi, do you address your problem? I have met the same problem here!

kpertsch, 2024-12-06:
hi -- unfortunately this memory increase is common for RLDS based dataloaders and we have also observed it in the large OpenVLA training runs. It's unfortunately hard to fix, since it's an issue with the underlying tensorflow libraries.
As a fix, you can try to reduce the number of data loading workers and threads here: https://github.com/openvla/openvla/blob/e2929881bb544d53b1a5b6c3e7d4bddd7fbf6bd9/prismatic/vla/datasets/datasets.py#L110

This will reduce the overall memory consumption, so while the usage will still increase over time, it will happen at a lower overall level, so hopefully will prevent crashing. The RAM usage does plateau at some point, but it looks like with your current config this point is higher than the available system memory.

closed 2024-12-06 by kpertsch