openvla/openvla #172: The seed in openvla/experiments/robot/libero/run_libero_eval.py has no effect on the eval result

Icecream-blue-sky, 2024-12-06:
I set seed 7 and seed 42 for libero evaluation. But the results is totally the same (I mean for every episode). And I find the seed won't change the initial state. So what should the seed take effect on?

kpertsch, 2024-12-06:
Libero retrieves the initial states from a fixed list to ensure reproducibility so this behavior is expected: https://github.com/openvla/openvla/blob/e2929881bb544d53b1a5b6c3e7d4bddd7fbf6bd9/experiments/robot/libero/run_libero_eval.py#L153

The seed variable is a "catch all" for any randomness in the code (that's controllable), but from your experiments it looks like there isn't any meaningful randomness atm.

closed 2024-12-06 by kpertsch

Icecream-blue-sky, 2024-12-09:
> Libero retrieves the initial states from a fixed list to ensure reproducibility so this behavior is expected: [...]

Yes, I conducted the experiment completely according to the steps in the paper. But there isn't any meaningful randomness. Why?
[image]

XilunZhangRobo, 2025-03-31:
Hi, 
I have the same question as @Icecream-blue-sky . Were you able to introduce some randomness to the model output?