openvla/openvla #175: About multi-node training RendezvousTimeoutError

SpaceLearner, 2024-12-10:
Hi OpenVLA Team,

It is really a nice work and I have learn't a lot. However, I encountered RendezvousTimeoutError when using 2 nodes with each node node 8 A100 GPUs for training. Do I need to modify the code to support multi-node multi-gpu training?  Here is my launching script.

`
export NCCL_DEBUG=INFO
torchrun --standalone   --master_addr=$MASTER_ADDR --master_port=$MASTER_PORT --nnodes=$WORLD_SIZE --nproc-per-node 8 --node_rank=$RANK vla-scripts/train_dva.py \
  --data_root_dir /root/data/workspace/workgroup/gjy/datasets \
  --run_root_dir /root/data/damovl/jiayan/checkpoints \
  --vla.type qwen2vl-2b-224px+mx-bridge \
  --image_aug True \
  --wandb_project DVA_Pretrain \
  --wandb_entity guojy001 \
  --save_interval 5000 \
  --is_resume False
`

closed 2024-12-10 by SpaceLearner

Akila-Ayanthi, 2025-01-20:
Hi @SpaceLearner were you able to sort this out?
I'm also getting the same error

Akila-Ayanthi, 2025-06-25:
Hi @SpaceLearner 
did you find a solution?