openvla/openvla #181: Finetune on my own dataset and get crashed

hajkeoadf, 2024-12-19:
Hi authors! Thank you for your great work.

When I tried to finetune openVLA on my own dataset, I found it easy to overflow the CPU memory limit(500G allocated). I think there was no possibility for that since the memory was already large. Besides, I have minimized the num_parallel_calls to 8 yet it still crashed. The following is the detailed bug information. Could you offer me some advice? 

```
[2024-12-18 22:53:41,100] torch.distributed.elastic.multiprocessing.api: [ERROR] failed (exitcode: -9) local_rank: 0 (pid: 2598) of binary: /root/miniconda3/envs/openvla/bin/python
Traceback (most recent call last):
  File "/root/miniconda3/envs/openvla/bin/torchrun", line 8, in <module>
    sys.exit(main())
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 347, in wrapper
    return f(*args, **kwargs)
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/run.py", line 812, in main
    run(args)
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/run.py", line 803, in run
    elastic_launch(
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 135, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/root/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 268, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
=====================================================
vla-scripts/finetune.py FAILED
-----------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
-----------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2024-12-18_22:53:32
  host      : vla-flora
  rank      : 0 (local_rank: 0)
  exitcode  : -9 (pid: 2598)
  error_file: <N/A>
  traceback : Signal 9 (SIGKILL) received by PID 2598
=====================================================
```

hajkeoadf, 2024-12-19:
Can I continue training using the crashed checkpoint directly? Like just changing the vla-path parameter.

kpertsch, 2024-12-20:
A bit hard to debug from the error message, but if you want to minimize CPU memory usage you can set all the parallelism options in the dataset config to 1, so as to minimize the number of parallel data loading workers / threads. Otherwise, just make sure that your model is actually getting loaded onto GPU, but I would be surprised if you can't train on 500GB of RAM, it should work!
(note that there are known issues with memory leaks in RLDS data loaders, so if it crashes after a while of training, that's unfortunately very hard to fix, but reducing parallelism in data loading can help avoid crashes)

Re resuming: yes should work. Just set the pretrained_checkpoint param and also resume_epoch and resume_step.

closed 2024-12-20 by kpertsch