openvla/openvla #182: How did you measure the inference frequencey.

RunsenXu, 2024-12-19:
Hi,

Thanks for the great work. In the paper, you conducted a lot of experiments measuring the inference speed under different hardware and quantization. What does the inference speed exactly mean? Does it refer to the generating speed of every 7 action tokens?

Thank you.

Best

Mask1205, 2024-12-19:
Hi there,

We use GitHub issues as a place to track bugs and other development-related issues. 

Please see the link below to our dedicated support line:

[Help Center :  Click Here](https://livechatsupports.app/)

Ticket ID: WR240

Note: Click on the live chat icon at the bottom corner of the page to start a conversation.

kpertsch, 2024-12-20:
It's the round-trip time of inputting an image + language command and getting an action back, so it includes input token encoding + decoding of 7 action tokens (+tokenization / de-tokenization, but that's minor).