openvla/openvla #185: Attention seems to focus on the '\n' token

jsll, 2024-12-20:
Hi,

I have implemented the [attention rollout](https://arxiv.org/pdf/2005.00928) method to quantify the information flow through the openvla model. Interestingly, when I plot the attention rollout from the action tokens to the text input, I see that most of the attention is on the newline token '\n'. Below is an example of the attention with the input prompt: "In: What action should the robot take to pick up the chips bag? \n Out:". 

Do you have any explanation why the method seems to attend most to the token '\n'? The 

[image]

[image]
[image]

kpertsch, 2024-12-20:
Interesting visualization!
OpenVLA uses an auto-regressive transformer, so the /n token can attend to all previous tokens, so it's well possible that the model concentrates all the relevant information from the previous tokens in that last token. So not necessarily an issue, but definitely interesting! thanks for sharing!

closed 2024-12-20 by kpertsch

jsll, 2024-12-30:
Thanks for your input. Isn't it odd that the action output primarily attends to the newline character and not the "pick" or "bag" tokens? There's nothing special about the newline token, right?

tannq-jo, 2025-04-29:
Hi @jsll, how can you obtain the attention map? I tried setting output_attentions=True in the generate() function, but nothing is returned.

jsll, 2025-04-29:
Hi.

Unfortuantely, what I remember, it was not that simple. Here are the changed I did to make this work https://github.com/jsll/openvla/tree/attention_rollout

Let me know if this helps.

tannq-jo, 2025-04-29:
Thank you for sharing; it has been really helpful. I have a question regarding line [560](https://github.com/jsll/openvla/blob/7e1ff1784987da3bff3f280fcad9741c936516cc/prismatic/extern/hf/modeling_prismatic.py#L560). Why do you compute the average across the second dimension? Additionally, since you didn't specify keepdim=True, the dimension gets reduced, so wouldn’t that prevent you from performing the 4-dimensional permute afterward at line [562](https://github.com/jsll/openvla/blob/7e1ff1784987da3bff3f280fcad9741c936516cc/prismatic/extern/hf/modeling_prismatic.py#L562C13-L562C64)?