openvla/openvla #187: Sharing Training Loss

LibertFan, 2024-12-27:
Could you share the training loss for better reproduce OpenVLA?

Thanks a lot!

siddk, 2025-02-03:
Yes! Note that all y-axes are on a log scale (with an exception of the action token accuracy graph). The blue line indicates the period of training prior to shifting the data mixture distribution, while purple reflects after. Please let use know if you have any other questions!

[image]
[image]
[image]

LukeLIN-web, 2025-02-21:
Why here is a big loss drop in 180k steps?

siddk, 2025-02-21:
Right - so like we mentioned in Section 3.3 of the paper, we changed the mixture of datasets we trained on for the last third of training (see screenshot below). That's the switch from blue to purple at 180K steps!

[image]

closed 2025-03-01 by moojink