openvla/openvla #189: Batch Inference

GuanxingLu, 2025-01-01:
Thanks for your great work to the community.

https://github.com/openvla/openvla/blob/1b024f242eda833dc8e321953f25cfd5f3d2f76d/prismatic/extern/hf/modeling_prismatic.py#L326

LukeLIN-web, 2025-02-06:
Why we need batch generation in robot case.

Akila-Ayanthi, 2025-02-09:
I'm also curious to know if you plan to support batch generation.

Qu3tzal, 2025-02-11:
> Why we need batch generation in robot case.

To do simulation, you can run multiple simulations in parallel and have the model run in batch mode on the observations.

GuanxingLu, 2025-05-31:
I've tried both left and right padding methods for batch generation, but the results are still incorrect. Additionally, I implemented OpenVLA using vLLM, and the batch decoding results from vLLM seem to be accurate (SR 83.60% bs=10 vs 83.40% bs=1 on LIBERO-Spatial). You can check out my implementation here: https://github.com/GuanxingLu/vlarl.

hnyu, 2025-05-31:
> I've tried both left and right padding methods for batch generation, but the results are still incorrect. Additionally, I implemented OpenVLA using vLLM, and the batch decoding results from vLLM [...]

You cannot do left padding just for inference because the positional encodings for the prompts are not seen during training. The finetuning itself has to be re-done.