openvla/openvla #195: How to accelerate LIBERO evaluation speed?

YunzeMan, 2025-01-13:
Thanks for the great effort! I'm wondering if anyone have experience on making LIBERO evaluation faster?

I am using an A100 machine with 128 CPUs and it takes about 10 hours to finish the 10*50=500 episodes of evaluation. (And there are 4 groups of tasks) I think launching multiple jobs in parallel is a direct way to save time. But is there any other direct ways to accelerate this process. I am new to LIBERO and not quite sure where the bottleneck usually is.

LukeLIN-web, 2025-02-04:
I am using an A6000.
In my case , it is 
  8%|█████████▎                                                                                                          | 4/50 [02:59<34:21, 44.82s/it]
34 mins 50 , yes it is slow.

buduz, 2025-02-26:
In fact, the evaluation of LIBERO does not require its dataset, as it is executed in a simulation environment.