openvla/openvla #196: Error LORA finetuning custom dataset

cian-shan, 2025-01-15:
Hi, I'm seeing the following error when adding and trying to run LORA finetuning on a custom test dataset. I've emboldened the errors below...

I have updated both the registration files - configs.py and transforms.py

**File "/export/home/cshanah/miniconda3/envs/openvla/lib/python3.10/site-packages/tensorflow_datasets/core/registered.py", line 301, in imported_builder_cls
    raise DatasetNotFoundError(f'Dataset {name} not found.')**
```
tensorflow_datasets.core.registered.DatasetNotFoundError: Dataset ur10_finetune not found.
Available datasets:
        - abstract_reasoning
        - accentdb
        - aeslc
        - aflw2k3d
        - ag_news_subset
        - ai2_arc
        - ai2_arc_with_ir
        - amazon_us_reviews
.....
        - youtube_vis

Check that:
    - if dataset was added recently, it may only be available
      in `tfds-nightly`
    - the dataset name is spelled correctly
    - dataset class defines all base class abstract methods
    - the module defining the dataset class is imported
```
**No builder could be found in the directory: /nfs/Data/asimov/Datasets for the builder: ur10_finetune.
No registered data_dirs were found in:
        - /nfs/Data/Datasets**
```
[2025-01-15 23:06:21,751] torch.distributed.elastic.multiprocessing.api: [ERROR] failed (exitcode: 1) local_rank: 0 (pid: 1850773) of binary: /export/home/cshanah2/miniconda3/envs/openvla/bin/python
Traceback (most recent call last):
  File "/export/home/cshanah/miniconda3/envs/openvla/bin/torchrun", line 8, in <module>
    sys.exit(main())
  File "/export/home/cshanah/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 347, in wrapper
    return f(*args, **kwargs)
  File "/export/home/cshanah/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/run.py", line 812, in main
    run(args)
  File "/export/home/cshanah/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/run.py", line 803, in run
    elastic_launch(
  File "/export/home/cshanah/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 135, in __call__

    return launch_agent(self._config, self._entrypoint, list(args))
  File "/export/home/cshanah/miniconda3/envs/openvla/lib/python3.10/site-packages/torch/distributed/launcher/api.py", line 268, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError:
============================================================
/export/home/cshanah/openvla/vla-scripts/finetune.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2025-01-15_23:06:21
  host      : adilk-srv-ai1
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 1850773)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html

```

LukeLIN-web, 2025-02-07:
Same when I using `modified_libero_rlds`

lostgreen, 2025-04-18:
@cian-shan Hi, have you solved the question? I met the same problem.

mxp1234, 2025-04-21:
> [@cian-shan](https://github.com/cian-shan) Hi, have you solved the question? I met the same problem.你好，你的问题解决了吗？我也遇到了同样的问题。

 I met the same problem and have no idea with that

lostgreen, 2025-04-21:
I solved this by changing the dir. First I use
```
  --data_root_dir /home/zhouxueyang/tensorflow_datasets/libero_object_with_trigger_mix \
  --dataset_name libero_object_with_trigger_mix \
```
and it raise the error `no builder found`,
then I change it like this:
```
  --data_root_dir /home/zhouxueyang/tensorflow_datasets \
  --dataset_name libero_object_with_trigger_mix \
```
it worked. I don't know whether you are meeting the same dir problem

YangJC112, 2025-05-17:
I've also created my own dataset and modified transform.py and config.py, but I encountered the same error. Have you solved it?

YangJC112, 2025-05-17:
@cian-shan  I've also created my own dataset and modified transform.py and config.py, but I encountered the same error. Have you solved it?