openvla/openvla #197: The performance in the LIBERO simulation is quite low

k-makihara, 2025-01-21:
I tried to evaluate the finetuned-OpenVLA in LIBERO. I just used the pipeline in README.md.
Why are the task mostly not worked? In LIBERO-Spaitial, the fine-tuned model ([openvla/openvla-7b-finetuned-libero-spatial](https://huggingface.co/openvla/openvla-7b-finetuned-libero-spatial)) performed all 15 episodes was failed.

Here, shows the script

`python experiments/robot/libero/run_libero_eval.py \
  --model_family openvla \
  --pretrained_checkpoint openvla/openvla-7b-finetuned-libero-spatial \
  --task_suite_name libero_spatial \
  --center_crop True`

Thank you,

LukeLIN-web, 2025-02-04:
In my case it is fine. 

I have LIBERO/libero/libero/../datasets does not exist!
But it still can run , and Saved rollout MP4 at path ./rollouts/2025_02_03/2025_02_03-19_06_36--episode=1--success=True--task=pick_up_the_black_bowl_between_the_plate_and_the_r.mp4 Success: True episodes completed so far: 1 successes: 1 (100.0%)
Why it can still run without dataset?

Qu3tzal, 2025-02-11:
@k-makihara I'm getting a similar problem. Towards the end of my training I reach a ~95% action accuracy on LIBERO-90 but evaluation give me all failed on the episodes I tried.

LukeLIN-web, 2025-02-24:
> [@k-makihara](https://github.com/k-makihara) I'm getting a similar problem. Towards the end of my training I reach a ~95% action accuracy on LIBERO-90 but evaluation gives me all failed on the episodes I tried.

I have same problem

LukeLIN-web, 2025-03-01:
Now they have https://github.com/moojink/openvla-oft  
13s per episode

k-makihara, 2025-05-11:
@LukeLIN-web @Qu3tzal Did you try the evaluation using the pre-trained model in Huggingface?
In my case, the problem was fixed. The reasons may depend on your GPU or CPU. When I use NVIDIA RTX A6000 Ada, the failure happens. But in the case of RTX A6000 or RTX 4090, and NVIDIA H200(in Cloud GPU), the evaluation was successfully done (about 83% in 12 trials). You may need to think about the hardware environment.

But I still don't know why the difference in the GPU causes the failure in LIBERO...