openvla/openvla #199: LIBERO Environment

MstarLioning, 2025-01-24:
Hello, when I tried to replicate the test using LIBERO's official guide, I encountered the following error:

(simpler_env_openvla) ch@4090-8:/home/lsf_storage/homes/ch/claude/openvla$ python experiments/robot/libero/run_libero_eval.py --model_family openvla --pretrained_checkpoint openvla/openvla-7b-finetuned-libero-spatial --task_suite_name libero_spatial --center_crop True
2025-01-24 10:57:23.210558: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-01-24 10:57:23.210629: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-01-24 10:57:23.211657: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-01-24 10:57:23.217776: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-01-24 10:57:24.518364: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
[robosuite WARNING] No private macro file found! (macros.py:53)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
[robosuite WARNING] To setup, run: python /home/ch/.conda/envs/simpler_env_openvla/lib/python3.10/site-packages/robosuite/scripts/setup_macros.py (macros.py:55)
[*] Instantiating Pretrained VLA model
[*] Loading in BF16 with Flash-Attention Enabled
Traceback (most recent call last):
  File "/home/lsf_storage/homes/ch/claude/openvla/experiments/robot/libero/run_libero_eval.py", line 297, in <module>
    eval_libero()
  File "/home/ch/.conda/envs/simpler_env_openvla/lib/python3.10/site-packages/draccus/argparsing.py", line 203, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/lsf_storage/homes/ch/claude/openvla/experiments/robot/libero/run_libero_eval.py", line 107, in eval_libero
    model = get_model(cfg)
  File "/home/lsf_storage/homes/ch/claude/openvla/experiments/robot/robot_utils.py", line 44, in get_model
    model = get_vla(cfg)
  File "/home/lsf_storage/homes/ch/claude/openvla/experiments/robot/openvla_utils.py", line 51, in get_vla
    if cfg.use_depth_model:
AttributeError: 'GenerateConfig' object has no attribute 'use_depth_model'
(simpler_env_openvla) ch@4090-8:/home/lsf_storage/homes/ch/claude/openvla$

LukeLIN-web, 2025-02-04:
I don't have this error

I have `LIBERO/libero/libero/../datasets does not exist!`
But it still can run , and `Saved rollout MP4 at path ./rollouts/2025_02_03/2025_02_03-19_06_36--episode=1--success=True--task=pick_up_the_black_bowl_between_the_plate_and_the_r.mp4
Success: True
# episodes completed so far: 1
# successes: 1 (100.0%)` 
Why it can still run without dataset?

Qu3tzal, 2025-02-11:
@LukeLIN-web stop spamming your question everywhere please. It can still run because LIBERO is a _simulated_ environment, it doesn't need a dataset.

MstarLioning, 2025-03-18:
@MikeYangsir Sorry, it's been a long time so I've kind of forgotten. However, I checked the previous error message and I faintly recall that it might have been caused by the transformers package version being too high.

MikeYangsir, 2025-03-18:
已收到！