openvla/openvla #200: OpenVLA usage with custom robot

GeraltRivskiy, 2025-01-24:
Hello! I am a beginner in robotics deep learning, I've examined this project but haven't seen any information on how to use this model with my robot (Hiwonder JetAuto Pro), how I should implement adapter from model output to robot actions or do I have to collect dataset with mine robot and fine-tune model? Any advices?

navono, 2025-03-24:
any progress?

GeraltRivskiy, 2025-03-24:
> any progress?

No, I was assigned to other task, but now I have to turn back to it
I saw nvidia gr00t that seems has slightly more detailed how-to documentation, but I didn't went through it yet.
Any advices on OpenVLA still would be much appreciated, since gr00t is more about humanoid robots and I have an arm on mobile platform

navono, 2025-03-25:
@GeraltRivskiy Thanks for response. I'll also take a look at nvidia gr00t.