openvla/openvla #203: Subject: Inquiry Regarding the Selection of unnorm_key Parameter in LIBERO

MstarLioning, 2025-01-26:
Dear Author,

Hello! Thank you very much for providing the LIBERO interface. However, I have a question that remains unclear to me: my task is to evaluate a custom dataset using OpenVLA, and I do not wish to perform post-training or fine-tuning with OpenVLA. At the same time, I still need to provide the unnorm_key parameter. When I use other fine-tuned models paired with different unnorm_key values (for example, LIBERO-spatial), the success rate of the robotic arm remains at 0%. However, when I use the original, unmodified OpenVLA, I find the available hyperparameters to choose from are:

['austin_buds_dataset_converted_externally_to_rlds', 'austin_sailor_dataset_converted_externally_to_rlds', 'austin_sirius_dataset_converted_externally_to_rlds', 'bc_z', 'berkeley_autolab_ur5', 'berkeley_cable_routing', 'berkeley_fanuc_manipulation', 'bridge_orig', 'cmu_stretch', 'dlr_edan_shared_control_converted_externally_to_rlds', 'dobbe', 'fmb_dataset', 'fractal20220817_data', 'furniture_bench_dataset_converted_externally_to_rlds', 'iamlab_cmu_pickup_insert_converted_externally_to_rlds', 'jaco_play', 'kuka', 'nyu_franka_play_dataset_converted_externally_to_rlds', 'roboturk', 'stanford_hydra_dataset_converted_externally_to_rlds', 'taco_play', 'toto', 'ucsd_kitchen_dataset_converted_externally_to_rlds', 'utaustin_mutex', 'viola']

Could you please advise me on how to select the appropriate unnorm_key in this situation? (I mean, how should I set this hyperparameter when I only want to perform evaluation without fine-tuning?) I look forward to your response. Thank you very much!

closed 2025-02-11 by MstarLioning

Luca-Wiehe, 2025-05-07:
I would still be interested in this. Which unnorm key did you use?

MstarLioning, 2025-05-08:
@Luca-Wiehe Hello.I try through almost all the unnorm_keys, and observed that as soon as an unnorm_key didn’t line up with the dataset, everything started to jitter wildly. After that, we skipped over this issue and chose SimplerEnv as our simulation environment. It now seems that we either have to retrain, or stick with a fixed number and format of samples. Hope this helps!