openvla/openvla #208: Multiple GPU load cause CPU OOM

LukeLIN-web, 2025-02-16:
When I run 6GPU ,

```
    torchrun --standalone --nnodes 1 --nproc-per-node 6 vla-scripts/train.py \
        --pretrained_checkpoint /prism-dinosiglip-224px+mx-bridge+n0+b1+x7/checkpoints/spatial.pt \
        --vla.type prism-dinosiglip-224px+mx-bridge \
        --is_resume False \
        --stage align
``` 

 Loading VLA prism-dinosiglip-224px+7b from Checkpoint     cause my 256GB CPU memory OOM. 
How to avoid it ?

LukeLIN-web, 2025-02-21:
My solution 
```
        # Load from Checkpoint (Custom --> should load both *projector* and *llm* weights)
        # model_state_dict = torch.load(pretrained_checkpoint, map_location="cpu")["model"]
        model_state_dict = torch.load(pretrained_checkpoint, map_location="cuda")["model"] # 
```

closed 2025-02-21 by LukeLIN-web

swstbecrpgmail, 2025-06-04:
> My solution
> 
> ```
>         # Load from Checkpoint (Custom --> should load both *projector* and *llm* weights)
>         # model_state_dict = torch.load(pretrained_checkpoint, map_location="cpu")["model"]
>         model_state_dict = torch.load(pretrained_checkpoint, map_location="cuda")["model"] # 
> ```
> 
> each process occupy 11.5GB at beginning. but finally each occupy 32GB makes my server OOM. This is because RLDS dataset [#4](https://github.com/openvla/openvla/issues/4)

Hello, have you solved this problem?