openvla/openvla #215: The details of fine-tuning Octo using LIBERO

buduz, 2025-02-26:
First of all, thank you very much for your excellent work! Recently, I have been trying to reproduce the experimental results of fine-tuning Octo using LIBERO as reported in your paper. However, the success rate in the evaluation always comes out as 0. I was wondering if you would be willing to share the finetune_config（the path is /octo/scripts/configs/finetune_config.py）. For example, did you use wrist images, proprioceptive information, etc., during the fine-tuning process? Are the config and transforms of LIBERO's RLDS consistent with the config of OpenVLA in your paper? Thank you once again!

Elycyx, 2025-03-27:
Hi, I’ve also been trying to fine-tune Octo on LIBERO and encountered the same issue with a 0% success rate. Have you found a solution?

buduz, 2025-03-27:
> Hi, I’ve also been trying to fine-tune Octo on LIBERO and encountered the same issue with a 0% success rate. Have you found a solution?

I still cannot reproduce the success rate in the paper, but I have attempted a configuration, and the final evaluation result is not 0%. You might want to try training without using wrist-view images.

Elycyx, 2025-03-27:
> I still cannot reproduce the success rate in the paper, but I have attempted a configuration, and the final evaluation result is not 0%. You might want to try training without using wrist-view images.

OK, I will try that, thank you for your reply.

Elycyx, 2025-03-29:
Hi, could you please share your `finetune_config.py`? I tried fine-tuning Octo again, but the results are still not good. I print Octo's output and find that it's always much smaller compared to OpenVLA's output, like `[-0.000 0.002 0.001 -0.002 0.000 0.003 0.002]
` which causes the robot in LIBERO to barely move.

buduz, 2025-03-29:
> Hi, could you please share your `finetune_config.py`? I tried fine-tuning Octo again, but the results are still not good. I print Octo's output and find that it's always much smaller compared to [...]

I actually didn't make many changes. You can give it a try, and I recommend using octo-small instead of octo-base.
```
    # Fill this in for your own dataset!

    # There should be two image keys
    # first image key should be the third-person view (None if not used)
    # and second image key should be the wrist view (None if not used)

    FINETUNING_KWARGS = {
        "name": "libero_object_no_noops",
        "data_dir": "/data/modified_libero_rlds",
        "image_obs_keys": {"primary": "image",  "wrist": None},
        # "depth_obs_keys": {"primary": "depth_static", "wrist": "depth_gripper"},
        # "proprio_obs_key": "proprio",
        "language_key": "language_instruction",
        "action_proprio_normalization_type": "normal",
        # We want to avoid normalizing the gripper
        "action_normalization_mask": [True, True, True, True, True, True, False],
        # standardize_fn is dynamically loaded from a file
        # for example: "experiments/kevin/custom_standardization_transforms.py:aloha_dataset_transform"
        "standardize_fn": ModuleSpec.create(
            "octo.data.oxe.oxe_standardization_transforms:libero_dataset_transform",
        ),
        # If the default data loading speed is too slow, try these:
        # "num_parallel_reads": 8,  # for reading from disk / GCS
        # "num_parallel_calls": 16,  # for initial dataset construction
    }

    if mode == "full":
        frozen_keys = None
    elif mode == "head_only":
        frozen_keys = ("octo_transformer.*",)
    elif mode == "head_mlp_only":
        frozen_keys = (
            "octo_transformer.*",
            "heads_*.map_head.probe",
            "heads_*.map_head.MultiHeadDotProductAttention_0.*",
        )
    else:
        raise ValueError("Invalid mode")

    max_steps = FieldReference(100000)
    window_size = FieldReference(default=1)

    config = dict(
        # pretrained_path=placeholder(str),
        pretrained_path='hf://rail-berkeley/octo-small-1.5',
        pretrained_step=placeholder(int),
        batch_size=64,
        shuffle_buffer_size=10000,
        num_steps=max_steps,
        log_interval=100,   # 100
        eval_interval=5000, # 5000
        save_interval=10000, # 5000
        # save_dir=placeholder(str),
        save_dir='./octo_libero/checkpoint',
        seed=42,
        wandb=dict(
            project="octo_libero", group=placeholder(str), entity=placeholder(str)
        ),
        dataset_kwargs=FINETUNING_KWARGS,
        modality=task,
        finetuning_mode=mode,
        window_size=window_size,
        optimizer=dict(
            learning_rate=dict(
                name="cosine",
                init_value=0.0,
                peak_value=3e-4,
                warmup_steps=2000,
                decay_steps=max_steps,
                end_value=0.0,
            ),
            weight_decay=0.01,
            clip_gradient=1.0,
            frozen_keys=frozen_keys,
            grad_accumulation_steps=None,  # if you are using grad accumulation, you need to adjust max_steps accordingly
        ),
        val_kwargs=dict(
            val_shuffle_buffer_size=1000,
            num_val_batches=16,
        ),
        viz_kwargs=dict(
            eval_batch_size=128,
            trajs_for_metrics=100,
            trajs_for_viz=8,
            samples_per_state=8,
        ),
    )

    if task == "image_conditioned":
        goal_relabeling_strategy = "uniform"
        keep_image_prob = 1.0
    elif task == "language_conditioned":
        goal_relabeling_strategy = None
        keep_image_prob = 0.0
    elif task == "multimodal":
        goal_relabeling_strategy = "uniform"
        keep_image_prob = 0.5
    else:
        raise ValueError("Invalid modality")

    traj_transform_kwargs = dict(
        window_size=window_size,
        action_horizon=4,
        goal_relabeling_strategy=goal_relabeling_strategy,
        task_augment_strategy="delete_task_conditioning",
        task_augment_kwargs=dict(
            keep_image_prob=keep_image_prob,
        ),
        # If the default data loading speed is too slow, try these:
        # num_parallel_calls=16,  # for less CPU-intensive ops
    )
    workspace_augment_kwargs = dict(
        random_resized_crop=dict(scale=[0.8, 1.0], ratio=[0.9, 1.1]),
        random_brightness=[0.1],
        random_contrast=[0.9, 1.1],
        random_saturation=[0.9, 1.1],
        random_hue=[0.05],
        augment_order=[
            "random_resized_crop",
            "random_brightness",
            "random_contrast",
            "random_saturation",
            "random_hue",
        ],
    )
    wrist_augment_kwargs = dict(
        random_brightness=[0.1],
        random_contrast=[0.9, 1.1],
        random_saturation=[0.9, 1.1],
        random_hue=[0.05],
        augment_order=[
            "random_brightness",
            "random_contrast",
            "random_saturation",
            "random_hue",
        ],
    )
    frame_transform_kwargs = dict(
        resize_size={
            "primary": (256, 256),  # workspace (3rd person) camera is at 256x256
            "wrist": (128, 128),  # wrist camera is at 128x128
        },
        image_augment_kwargs=dict(
            primary=workspace_augment_kwargs,
            wrist=wrist_augment_kwargs,
        ),
    )
    # If the default data loading speed is too slow, try these:
    config[
        "frame_transform_threads"
    ] = 16  # for the most CPU-intensive ops (decoding, resizing, augmenting)

    config["traj_transform_kwargs"] = traj_transform_kwargs
    config["frame_transform_kwargs"] = frame_transform_kwargs
    return ConfigDict(config)
```

Elycyx, 2025-03-29:
Thank you! I have been using Octo-Base, so I'll try Octo-Small instead.
By the way, how did you implement `octo.data.oxe.oxe_standardization_transforms:libero_dataset_transform`?

buduz, 2025-03-29:
> Thank you! I have been using Octo-Base, so I'll try Octo-Small instead. By the way, how did you implement `octo.data.oxe.oxe_standardization_transforms:libero_dataset_transform`?

Just follow treating processes of OpenVLA 
```
def libero_dataset_transform(trajectory: Dict[str, Any]) -> Dict[str, Any]:
    # gripper action is in -1 (open)...1 (close) --> clip to 0...1, flip --> +1 = open, 0 = close
    gripper_action = trajectory["action"][:, -1:]
    gripper_action = invert_gripper_actions(tf.clip_by_value(gripper_action, 0, 1))
    
    trajectory["action"] = tf.concat(
        [
            trajectory["action"][:, :6],
            gripper_action,
        ],
        axis=1,
    )
    # trajectory["observation"]["EEF_state"] = trajectory["observation"]["state"][:, :6]
    # trajectory["observation"]["gripper_state"] = trajectory["observation"]["state"][:, -2:]  # 2D gripper state
    
    trajectory["observation"]["proprio"] = tf.concat(
        (
            trajectory["observation"]["state"][:, :6],
            trajectory["observation"]["state"][:, -2:],
        ),
        axis=-1,
    )
    return trajectory
```

Yulie00, 2025-04-13:
> Hi, could you please share your `finetune_config.py`? I tried fine-tuning Octo again, but the results are still not good. I print Octo's output and find that it's always much smaller compared to [...]

Hello, could you please let me know if you were able to finetune Octo on Libero?

Elycyx, 2025-04-14:
> Hello, could you please let me know if you were able to finetune Octo on Libero?

Yes, I can. But even after training for 100k steps, the success rate is still only around 68%.

Yulie00, 2025-04-15:
> Yes, I can. But even after training for 100k steps, the success rate is still only around 68%.

Could you please share the scripts used for fine-tuning? I will try to improve the result

Elycyx, 2025-04-15:
> Could you please share the scripts used for fine-tuning? I will try to improve the result

Hi, the `finetune_config.py` I used was the same as the one shared by Buduz, and I didn't specifically tune any parameters. You might try printing Octo's output and watching the rollouts to debug the issue.

farewillwell, 2025-04-15:
> Hi, the `finetune_config.py` I used was the same as the one shared by Buduz, and I didn't specifically tune any parameters. You might try printing Octo's output and watching the rollouts to debug [...]

I noticed that when fine-tuning the model (octo-base), it doesn't seem to converge even after 100k steps. May I ask if you were using the octo-small variant instead? I'm also curious—would you mind sharing what your training curves looked like? Thank you!

Elycyx, 2025-04-15:
> I noticed that when fine-tuning the model (octo-base), it doesn't seem to converge even after 100k steps. May I ask if you were using the octo-small variant instead? I'm also curious—would you mind [...]

I previously tried training octo-base, but found that the performance wasn’t great, so now I’m training octo-small instead. Here's the curve of my training loss.

[image]

BTW, I’ll try training for more steps or tweaking octo-base when I have more time.

aliothz, 2025-06-04:
> > > > > Hi, could you please share your `finetune_config.py`? I tried fine-tuning Octo again, but the results are still not good. I print Octo's output and find that it's always much smaller compared to OpenVLA's output, like `[-0.000 0.002 0.001 -0.002 0.000 0.003 0.002] ` which causes the robot in LIBERO to barely move.
> > > > 
> > > > 
> > > > Hello, could you please let me know if you were able to finetune Octo on Libero?
> > > 
> > > 
> > > Yes, I can. But even after training for 100k steps, the success rate is still only around 68%.
> > 
> > 
> > Hi, could you please tell me which evaluation you used to get this 68% success rate? I tried the rollout_callback in octo/scripts/finetune.py but it seems not work (ERROR code: cannot recognize key unnormalization_statistics), and also refer to eval.py in https://github.com/niejnan/OpenVLA but also gives 0% success rate. Are you refering to anything in this repo or any other please? And could you please share the scripts for evaluation please?
> 
> I only used the evaluation script `run_libero_eval_octo.py` provided in the official OpenVLA library, but I implemented functions related to Octo in `openvla_util.py`, for example:
> 
> ```
> def get_octo(cfg):
>     """Loads and returns an Octo model from checkpoint."""
>     octo = OctoModel.load_pretrained(cfg.pretrained_checkpoint)
>     return octo
> 
> 
> def get_octo_action(octo, obs, task_label, unnorm_key, center_crop=False):
>     """Generates an action with the Octo policy (currently only supports one-step inference).
>     
>     Args:
>         octo: OctoModel instance
>         obs: observation dict containing 'full_image'
>         task_label: string describing the task
>         unnorm_key: key for unnormalization statistics
>         center_crop: whether to apply center crop augmentation
>     """
>     # Convert observation to the format expected by Octo
>     img = obs["full_image"]
>     if center_crop:
>         batch_size = 1
>         crop_scale = 0.9
> 
>         # Convert to TF Tensor and record original data type (should be tf.uint8)
>         img = tf.convert_to_tensor(img)
>         orig_dtype = img.dtype
> 
>         # Convert to data type tf.float32 and values between [0,1]
>         img = tf.image.convert_image_dtype(img, tf.float32)
> 
>         # Crop and then resize back to original size
>         img = crop_and_resize(img, crop_scale, batch_size)
> 
>         # Convert back to original data type
>         img = tf.clip_by_value(img, 0, 1)
>         img = tf.image.convert_image_dtype(img, orig_dtype, saturate=True)
> 
>         # Convert back to numpy array
>         img = img.numpy()
> 
>     observations = {
>         "image_primary": np.array(img)[None, None],  # Add batch and history dimensions
>         "timestep_pad_mask": np.array([[True]]),  # No padding
>         "proprio": np.array(obs['state'])[None, None]  # Reshape proprio to have batch and history dimensions
>     }
>     
>     # Create task specification
>     task = octo.create_tasks(texts=[task_label])
>     
>     # Get action from model
>     action = octo.sample_actions(
>         observations=observations,
>         tasks=task,
>         unnormalization_statistics=octo.dataset_statistics["action"],
>         rng=jax.random.PRNGKey(0)  # Fixed seed for deterministic behavior
>     )
>     
>     # Remove batch dimension and return
>     return action[0][0]
> ```

Thanks a lot! The problem should be on proprio, I will have a try on this!

aliothz, 2025-06-05:
> I only used the evaluation script `run_libero_eval_octo.py` provided in the official OpenVLA library, but I implemented functions related to Octo in `openvla_util.py`, for example:
>
> ```
> def [...]

Have you used proprio in finetuneing please? It works, but the performance is just 40% for me, I think maybe because I have not used proprio for finetuning?

aliothz, 2025-06-05:
> > > > > > > Hi, could you please share your `finetune_config.py`? I tried fine-tuning Octo again, but the results are still not good. I print Octo's output and find that it's always much smaller compared to OpenVLA's output, like `[-0.000 0.002 0.001 -0.002 0.000 0.003 0.002] ` which causes the robot in LIBERO to barely move.
> > > > > > 
> > > > > > 
> > > > > > Hello, could you please let me know if you were able to finetune Octo on Libero?
> > > > > 
> > > > > 
> > > > > Yes, I can. But even after training for 100k steps, the success rate is still only around 68%.
> > > > 
> > > > 
> > > > Hi, could you please tell me which evaluation you used to get this 68% success rate? I tried the rollout_callback in octo/scripts/finetune.py but it seems not work (ERROR code: cannot recognize key unnormalization_statistics), and also refer to eval.py in https://github.com/niejnan/OpenVLA but also gives 0% success rate. Are you refering to anything in this repo or any other please? And could you please share the scripts for evaluation please?
> > > 
> > > 
> > > I only used the evaluation script `run_libero_eval_octo.py` provided in the official OpenVLA library, but I implemented functions related to Octo in `openvla_util.py`, for example:
> > > ```
> > > def get_octo(cfg):
> > >     """Loads and returns an Octo model from checkpoint."""
> > >     octo = OctoModel.load_pretrained(cfg.pretrained_checkpoint)
> > >     return octo
> > > 
> > > 
> > > def get_octo_action(octo, obs, task_label, unnorm_key, center_crop=False):
> > >     """Generates an action with the Octo policy (currently only supports one-step inference).
> > >     
> > >     Args:
> > >         octo: OctoModel instance
> > >         obs: observation dict containing 'full_image'
> > >         task_label: string describing the task
> > >         unnorm_key: key for unnormalization statistics
> > >         center_crop: whether to apply center crop augmentation
> > >     """
> > >     # Convert observation to the format expected by Octo
> > >     img = obs["full_image"]
> > >     if center_crop:
> > >         batch_size = 1
> > >         crop_scale = 0.9
> > > 
> > >         # Convert to TF Tensor and record original data type (should be tf.uint8)
> > >         img = tf.convert_to_tensor(img)
> > >         orig_dtype = img.dtype
> > > 
> > >         # Convert to data type tf.float32 and values between [0,1]
> > >         img = tf.image.convert_image_dtype(img, tf.float32)
> > > 
> > >         # Crop and then resize back to original size
> > >         img = crop_and_resize(img, crop_scale, batch_size)
> > > 
> > >         # Convert back to original data type
> > >         img = tf.clip_by_value(img, 0, 1)
> > >         img = tf.image.convert_image_dtype(img, orig_dtype, saturate=True)
> > > 
> > >         # Convert back to numpy array
> > >         img = img.numpy()
> > > 
> > >     observations = {
> > >         "image_primary": np.array(img)[None, None],  # Add batch and history dimensions
> > >         "timestep_pad_mask": np.array([[True]]),  # No padding
> > >         "proprio": np.array(obs['state'])[None, None]  # Reshape proprio to have batch and history dimensions
> > >     }
> > >     
> > >     # Create task specification
> > >     task = octo.create_tasks(texts=[task_label])
> > >     
> > >     # Get action from model
> > >     action = octo.sample_actions(
> > >         observations=observations,
> > >         tasks=task,
> > >         unnormalization_statistics=octo.dataset_statistics["action"],
> > >         rng=jax.random.PRNGKey(0)  # Fixed seed for deterministic behavior
> > >     )
> > >     
> > >     # Remove batch dimension and return
> > >     return action[0][0]
> > > ```
> > 
> > 
> > Have you used proprio in finetuneing please? It works, but the performance is just 40% for me, I think maybe because I have not used proprio for finetuning?
> 
> Yeah, I used proprio in fine-tuning. However, I find that whether or not I use proprio during evaluation, the results don’t change much.

Thank you! By the way, in my case the model seems a bit overfit if I look at the rollout video, it's sometimes doing some other tasks (the task is 'open the middle drawer of the cabinet' but the model is doing 'put the wine bottle on the rack'), maybe you can try for some intermediate states (like 80000 or 90000) to have better performance if you are still working on this.

lFelixC, 2025-07-04:
@Elycyx Hi! Thank you for providing the octo_util code. I have a quick question regarding the line `img = crop_and_resize(img, crop_scale, batch_size)`. Given that the input image size is 224x224 pixels, but Octo requires 256x256, I'm wondering if this crop_and_resize function handles the image resizing as well? Thanks for your help!