openvla/openvla #218: About OpenVLA-OFT performance on LIBERO task suites

georgegu1997, 2025-02-28:
Hi, 

Thanks for the great work and open-sourcing OpenVLA and OpenVLA-OFT! They are very helpful to the community!

I saw there is no "issues" section on the OFT repo and thus am asking here:

Is the LIBERO performance of OpenVLA-OFT obtained by finetuning one model on each task suite? If so, why not finetuning a model on all task suites? Also, I think pi-0 uses only one model for all LIBERO task suites, and seems this is not noted in the current OFT paper.

moojink, 2025-03-01:
Hi @georgegu1997, thank you for letting us know! The Issues section in the openvla-oft repo should be open now.

Yes, you are right: we trained one OpenVLA-OFT policy per suite, following what was done in the original OpenVLA paper, so that we could fairly compare to the vanilla fine-tuned OpenVLA results (which was our main reference point in the “VLA fine-tuning design decisions study” portion of our OFT paper).

Your last point is correct. The last few rows of the LIBERO results table where we compare with other state-of-the-art methods is more uncontrolled. There we just wanted to compare with the absolute best known results from other methods. We can add a note to the paper to clarify this in our next revision on arXiv. Also, we can try fine-tuning on all task suites together later if we have time.

Thank you for the questions!

-Moo Jin

georgegu1997, 2025-03-01:
Thanks for the fast response! @moojink 

It would be great to have the OFT results of one model finetuned on all task suites. I think it should also perform very well, and I'm curious about the actual results.

georgegu1997, 2025-04-10:
Closing this issue. Seems it has been further followed up in https://github.com/moojink/openvla-oft/issues/21

closed 2025-04-10 by georgegu1997

moojink, 2025-04-28:
Hi @georgegu1997, please see https://github.com/moojink/openvla-oft/issues/21 for results with the OpenVLA-OFT checkpoint trained on all four LIBERO task suites combined. We also recently released the corresponding checkpoint and added instructions for using it. It should be very easy to get started!