openvla/openvla #219: Memory leak in full-finetuning

tyler-ingebrand, 2025-03-03:
When running the command in the README for full finetuning, there is a memory leak for CPU memory. See the figure below:

[image]

Have you run into this? Is there a known fix? After spending a few days debugging, it seems like the memory leak is inside of tensorflow datasets, but I am struggling to track it down further.

healenrens, 2025-04-15:
I have the same problem. And I think it might because prismatic or tensorflow dataset does not release previously loaded data during the training process, I am also looking for a solution.

mikygit, 2025-05-21:
Same here but with LoRa.
Did you manage to track it down further?