openvla/openvla #225: Try on LIBERO

MstarLioning, 2025-03-10:
Hello, thank you for your work! I encountered the following issues when trying to predict in a custom new scenario using OpenVLA:

For instance, when executing pick_up_the_black_bowl_from_table_center_and_place_it_on_the_plate, it works with 90% accuracy. However, if I change the file name by replacing "plate" with "stove" at the end, the model still executes it with 100% accuracy. What could be the reason for this?

When I reduce the filename of pick_up_the_black_bowl_from_table_center_and_place_it_on_the_plate to just pick up the black bowl, it fails to execute the command and instead causes the robotic arm to jitter randomly at the starting position.

When unnorm_key is set to spatial and I run the cases within the object category, the accuracy is 0%.

MstarLioning, 2025-03-11:
https://github.com/user-attachments/assets/f4821305-8b28-4beb-8eab-75a90c8d3c51

pick up the cookies

https://github.com/user-attachments/assets/96309b02-d8ef-4c43-b7a7-4743bc418acf

open_the_middle_drawer_of_the_cabinet

briliantnugraha, 2025-03-17:
Hi @MstarLioning, have you solved the issue? 
We encountered similar problem in here, where it basically moves randomly given simple prompts.

MstarLioning, 2025-03-17:
@briliantnugraha I'm sorry, we haven't solved this problem yet. Based on our controlled experiments, it seems that the only difference might be caused by the bias in unnorm_key. However, I don't know how to fix it. May I ask what approaches you have tried to validate this issue?

briliantnugraha, 2025-03-17:
Hi @MstarLioning , we're still fiddling with the simulator environment, haven't tried yet with real UR5 though (afraid of it wreaking havoc xD).

But Libero dataset has different arm robot, which "might" contribute to unstable/high failure results (see ref. in the bridge dataset).

MstarLioning, 2025-03-17:
@briliantnugraha It is indeed very strange—a slight change to the object results in random movement, yet in an imitation learning context it can operate at the accuracy reported in the paper. I'll check the issues later to see if anyone has asked or resolved a similar problem to ours.

MstarLioning, 2025-03-17:
@briliantnugraha By the way, may I ask how the pruned_init_ trajectory is generated in LIBERO? I simply selected a random trajectory. How do you handle it when defining new bddl files? I'm wondering if this could be one of the reasons for this result.

Reference: https://github.com/openvla/openvla/issues/138

briliantnugraha, 2025-03-17:
Hi @MstarLioning, unfortunately, we didn't proceed to test/fine-tuned using LIBERO dataset, since we want to "generate" what the authors did in the paper.

> Hi [@MstarLioning](https://github.com/MstarLioning), have you solved the issue? We encountered similar problem in here, where it basically moves randomly given simple prompts.

Unfortunately, despite testing with similar environment in the simulator, we still get **random movements**. I feel like we indeed might miss some config/setting in here, which I hope the authors could help us with.

MstarLioning, 2025-04-01:
@lostgreen you can refer to this ISSUE：https://github.com/Lifelong-Robot-Learning/LIBERO/issues/27