openvla/openvla #235: prompt to enter a new task, but libero still executes the original task

mengfanyu1991, 2025-03-27:


LukeLIN-web, 2025-03-27:
Hi, How you change it? Could you show your code?

mengfanyu1991, 2025-03-28:
> Hi, How you change it? Could you show your code?

https://github.com/user-attachments/assets/01019502-fd92-4180-9842-09a22848ffda
for example: pick_up_the_orange_juice_and_place_it_in_the_basket

LukeLIN-web, 2025-03-28:
> > Hi, How you change it? Could you show your code?
> 
> https://github.com/user-attachments/assets/01019502-fd92-4180-9842-09a22848ffda for example: pick_up_the_orange_juice_and_place_it_in_the_basket

I am not sure, One reason maybe https://github.com/openvla/openvla/blob/c8f03f48af692657d3060c19588038c7220e9af9/prismatic/vla/datasets/datasets.py#L62 
they don't consider language task?