openvla/openvla #239: Requires start <= limit when delta > 0: 0/-2 Bug

LukeLIN-web, 2025-04-08:
When I train on bridge dataset,
I also meet tensorflow.python.framework.errors_impl.InvalidArgumentError: {{function_node __wrapped__IteratorGetNext_output_types_22_device_/job:localhost/replica:0/task:0/device:CPU:0}} Requires start <= limit when delta > 0: 0/-2 Bug, before I decrease shuffle buffer to 90k . It is solved.

But when I train in   --dataset_name fractal20220817_data  
It happens again and cannot solve.

```
2025-04-07 19:03:12.384658: W tensorflow/core/framework/op_kernel.cc:1839] OP_REQUIRES failed at broadcast_to_op.cc:56 : INVALID_ARGUMENT: Dimension -2 must be >= 0
2025-04-07 19:03:12.384749: W tensorflow/core/framework/op_kernel.cc:1839] OP_REQUIRES failed at constant_op.cc:174 : INVALID_ARGUMENT: Dimension -2 must be >= 0
2025-04-07 19:03:12.384845: W tensorflow/core/framework/op_kernel.cc:1839] OP_REQUIRES failed at broadcast_to_op.cc:56 : INVALID_ARGUMENT: Dimension -2 must be >= 0
   raise core._status_to_exception(e) from None  # pylint: disable=protected-access
tensorflow.python.framework.errors_impl.InvalidArgumentError: {{function_node __wrapped__IteratorGetNext_output_types_22_device_/job:localhost/replica:0/task:0/device:CPU:0}} Requires start <= limit when delta > 0: 0/-2
         [[{{node range_4}}]] [Op:IteratorGetNext] name: 
2025-04-07 19:03:12.400790: W tensorflow/core/framework/op_kernel.cc:1839] OP_REQUIRES failed at broadcast_to_op.cc:56 : INVALID_ARGUMENT: Dimension -2 must be >= 0
2025-04-07 19:03:12.400959: W tensorflow/core/framework/op_kernel.cc:1839] OP_REQUIRES failed at broadcast_to_op.cc:56 : INVALID_ARGUMENT: Dimension -2 must be >= 0
2025-04-07 19:03:12.401009: W tensorflow/core/framework/op_kernel.cc:1839] OP_REQUIRES failed at constant_op.cc:174 : INVALID_ARGUMENT: Dimension -2 must be >= 0

```

LukeLIN-web, 2025-04-08:
Seems dataload have problem, but no clue
```
 File "/home/user1/anaconda3/envs/vlaoft/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 631, in __next__
    data = self._next_data()
  File "/home/user1/anaconda3/envs/vlaoft/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 675, in _next_data
    data = self._dataset_fetcher.fetch(index)  # may raise StopIteration
  File "/home/user1/anaconda3/envs/vlaoft/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 32, in fetch
    data.append(next(self.dataset_iter))
```

jasper0314-huang, 2025-04-09:
@LukeLIN-web  Are you using the openvla-oft codebase?

LukeLIN-web, 2025-04-09:
> [@LukeLIN-web](https://github.com/LukeLIN-web) Are you using the openvla-oft codebase?

Yes

jasper0314-huang, 2025-04-10:
> > [@LukeLIN-web](https://github.com/LukeLIN-web) Are you using the openvla-oft codebase?
> 
> Yes

I found that the issue occurs in the `chunk_act_obs` function when effective_traj_len <= 0, so I added this line:
`dataset = dataset.filter(lambda traj: tf.shape(traj["action"])[0] > future_action_window_size)`
before applying traj_transforms.chunk_act_obs. It seems that no errors occur now.

LukeLIN-web, 2025-04-10:
> I found that the issue occurs in the `chunk_act_obs` function when effective_traj_len <= 0, so I added this line: `dataset = dataset.filter(lambda traj: tf.shape(traj["action"])[0] > [...]

Amazing! You save my day! Thank you!

closed 2025-04-12 by LukeLIN-web