openvla/openvla #241: Failed to run command of finetune.py

huangguangheng, 2025-04-12:
I use command:
torchrun --standalone --nnodes 1 --nproc-per-node 8 vla-scripts/train.py \
  --pretrained_checkpoint <PATH TO openvla/openvla-7b-prismatic CHECKPOINT FILE: step-295000-epoch-40-loss=0.2200.pt> \
  --vla.type prism-dinosiglip-224px+mx-bridge \
  --data_root_dir <PATH TO BASE DATASETS DIR> \
  --run_root_dir <PATH TO LOG/CHECKPOINT DIR> \
  --run_id <OPTIONAL RUN ID FOR WANDB LOGGING> \
  --image_aug <True or False> \
  --wandb_project <PROJECT> \
  --wandb_entity <ENTITY> \
  --save_interval <NUMBER OF GRADIENT STEPS PER CHECKPOINT SAVE> \
  --is_resume False
and correctly configured various parameters, why is the error that dataset not included in the registered dataset of TensorFlow??
The bridge dataset was properly downloaded...

huangguangheng, 2025-04-12:
I completely follow the README.md document to perform....

closed 2025-04-12 by huangguangheng

103240, 2025-06-13:
> 我完全按照 README.md 文件来执行......

请问你解决了吗