openvla/openvla #242: fmb and fmb_dataset

shiyang1983, 2025-04-13:
Hi, https://github.com/openvla/openvla/blob/c8f03f48af692657d3060c19588038c7220e9af9/prismatic/vla/datasets/rlds/oxe/mixtures.py#L135 uses fmb_dataset but https://github.com/moojink/rlds_dataset_mod/blob/ad83e6c0efad5823540c0f6d3a05529596ead0b5/prepare_open_x.sh#L56 uses fmb. Is this a bug?

Thanks very much for your reply in advance!

JaehongJaehongMin, 2025-04-22:
Thank you for sharing your great work!
I think OpenVLA is one of the greatest works in this VLA era.

By the way, I have a somewhat related question with this issue.
I've just downloaded and preprocessd the OXE dataset with your custom code explained in README.
Almost of the whole dataset have been downloaded but fmb.

The whole size of the fmb dataset is 356GB in OXE spreadsheet (https://docs.google.com/spreadsheets/d/1rPBD77tk60AEIGZrGSODwyyzs5FgCU9Uz3h-3_t2A9g/edit?gid=0#gid=0).
But the downloaded (and preprocessed) fmb dataset consists over than 1.2TB which is 4 times bigger than original data.

I've read your comment on issue #101 that only a partial dataset of fmb is uploaded in OpenX GCP bucket.
But I think 4 times bigger is too much bigger.

JaehongJaehongMin, 2025-04-22:
Thank you again for sharing such impressive work!
I believe OpenVLA represents one of the most outstanding contributions in the VLA era.

I have a question related to the OXE dataset.
Following the instructions in the README, I downloaded and preprocessed the dataset using your custom code. Most of the datasets were processed successfully, except for the fmb subset.

According to the OXE spreadsheet (https://docs.google.com/spreadsheets/d/1rPBD77tk60AEIGZrGSODwyyzs5FgCU9Uz3h-3_t2A9g/edit?gid=0#gid=0), the total size of the fmb dataset is listed as 356GB.
However, after downloading and preprocessing, the resulting size exceeds 1.2TB, which is more than four times larger than the original. (Other datasets are almost same or smaller than raw datasets)

I came across your comment in issue https://github.com/openvla/openvla/issues/101 that only a partial portion of fmb is available in the OpenX GCP bucket. That said, the size difference seems unusually large.
May I kindly ask if this is expected behavior, or if there might be something I'm missing in the preprocessing step?

Thank you in advance for your help!

[image]