openvla/openvla #243: Question in transfering Libero to rlds

lostgreen, 2025-04-14:
Hi， thanks for such a great work. I made some data using Libero, but I met some problems in transforming to rlds. In the collected hdf5 file, there are many keys, but I'm not sure which to save. And `actions` `robot_states` shape is different from the rlds transform example. Could you offer your transformation code on Libero dataset? Thanks a lot.
```
/data/demo_0/actions
/data/demo_0/dones
/data/demo_0/obs/agentview_rgb
/data/demo_0/obs/ee_ori
/data/demo_0/obs/ee_pos
/data/demo_0/obs/ee_states
/data/demo_0/obs/eye_in_hand_rgb
/data/demo_0/obs/gripper_states
/data/demo_0/obs/joint_states
/data/demo_0/rewards
/data/demo_0/robot_states
/data/demo_0/states
```

closed 2025-04-14 by lostgreen

103240, 2025-04-23:
我也有这个问题，请问如何将hdf5转换成tfds

HaochenZ11, 2026-01-16:
Hi, would you mind sharing how you did the conversion?