openvla/openvla #246: Action tokenizer issues when reconstructing action

hanyangclarence, 2025-04-21:
Thanks for your great work! 

I am working with your action tokenizer, and encountered an issue. I see that your action tokenizer tokenizes action into ids of rare language tokens. Then these ids are decode to strings of rare characters. But when I re-encode these strings into ids, I found that the ids are different (some rare character are encoded as normal characters).

Here is my script. I change to a smaller model to allow running on my local wsl.
```python
import numpy as np

from prismatic.models import load
from prismatic.vla.action_tokenizer import ActionTokenizer

vlm = load("model/prism-qwen25-extra-dinosiglip-224px-0_5b")

action_tokenizer = ActionTokenizer(vlm.llm_backbone.get_tokenizer())

action = np.random.rand(4, 8)
action = (action - action.min()) / (action.max() - action.min()) * 2 - 1

# 2d ndarray to list of str
out = action_tokenizer(action)
# list of str to list of list of int
ids = action_tokenizer.tokenizer(out)["input_ids"]

print(action_tokenizer.tokenizer_len)
print(ids)
print(out)

# list of list of int to 2d ndarray
action_recon = np.array([action_tokenizer.decode_token_ids_to_actions(np.array(id)) for id in ids])

print()
print(action)
print()
print(action_recon)
```

The following screen shots are two of my test case. Since the vocab sizeof the tokenizer is over 150k, then the ids of the action tokens should also be values close to 150k. However in the first screen shot, after re-encoding the string into ids, I got ids that are around 99k.
[image]

Furthermore, in the second screenshot, after re-encoding, one string is encoded into 9 tokens.
[image]

So I would like to know what could be the issue? Is it that I am loading the tokenizer wrong? Or is it the issue with this version of tokenizer?

hanyangclarence, 2025-04-22:
I remembered that they said that their original tokenizer can only add a maximum of 256 extra tokens. So instead they choose to use the least frequently used text tokens to directly encode action. I followed such implementation  and ran into the bug I reported. 

I'm not sure whether it is because I am running in WSL, so some of the rare characters will be automatically changed to some more frequent character so that it can be displayed on screen or stored in Windows system. It's just like if you write a code in Windows, you can't directly run it in Linux due to incompatible new-line characters, and that you have to convert it with dos2unix. But I'm not sure, it's just my guess.

But anyway, I found that the tokenizer can indeed add more than 256 extra tokens, so I change to tokenizer.add_tokens() to solve the problem.

MyhooSalsa, 2025-04-22:
@hanyangclarence Thank you for your serious reply. Wish you success in your research.

hanyangclarence, 2025-05-05:
No but I found this error persists with the original openvla model with the class `transformers.models.llama.tokenization_llama_fast.LlamaTokenizerFast` tokenizer. In the `__call__` function in `prismatic.vla.datasets.datasets.RLDSBatchTransform`, I tried the following on my linux server:

```
(Pdb) action
array([ 0.5625    , -0.23418802,  0.7114178 , -0.34986407, -0.14421439,
        0.18722141,  0.        ], dtype=float32)
(Pdb) tokenized_action = self.action_tokenizer(action)
(Pdb) raw_action_tokens = self.base_tokenizer(tokenized_action)["input_ids"]
(Pdb) self.action_tokenizer.decode_token_ids_to_actions(np.array(raw_action_tokens))
array([ 0.99607843,  0.99607843,  0.56470588, -0.23529412,  0.71372549,
       -0.35294118, -0.14117647,  0.18823529,  0.        ])
```

where the reconstructed action is different from the original action.

Or more specificly, as shown in the following:
```
(Pdb) tokenized_action = self.action_tokenizer(action)
(Pdb) tokenized_action
'ܝ米씨මɵ拳Ÿ'
(Pdb) len(tokenized_action)
7
(Pdb) self.base_tokenizer(tokenized_action)["input_ids"]
[1, 29871, 31800, 31902, 31781, 31917, 31890, 31848, 31872]
```
The problem lies in the convertion from string to ids.

So is this expected/normal?

MyhooSalsa, 2025-05-06:
Hi @hanyangclarence . I have found that "input_ids" seems to be truncated to match action_dim in original openvla codes, you can check this file: _prismatic/models/vlas/openvla.py_ **line 90** . As you previous experiment above, I think it is normal for action error after encode-decode, but the action dim must be consistent. I am not sure whether my answer is you want.

closed 2025-05-06 by hanyangclarence