openvla/openvla #252: Question on OpenVLA mini

lostgreen, 2025-05-02:
Hi, how can i eval openvla-mini? I notice that only pretrain and train scripts are availabel.

lostgreen, 2025-07-25:
Sorry, we still did not figure out how to use this model, we use another vla instead_

Akila-Ayanthi, 2025-07-29:
I used model_family=prismatic and in the get_prismatic_vla_action() used the line
action, ids = vla.predict_action(processed_images, task_label, unnorm_key=unnorm_key, max_new_tokens=vla.get_action_dim(unnorm_key), **kwargs).

This gave better results but still not the numbers reported. 
Without max_new_tokens=vla.get_action_dim(unnorm_key), it was way worse, <5%.

ylhaichen, 2025-09-05:
> [@ylhaichen](https://github.com/ylhaichen) Yes, so can we evaluate without converting?

I don't think so. A conversion seems needed but the script wasn't provided.