openvla/openvla #253: When I encountered issues with the inference of the pre-trained model.

xyl9q, 2025-05-11:
When I run the following code

from transformers import AutoModelForVision2Seq, AutoProcessor
from PIL import Image
import torch

# 加载处理器与VLA
processor = AutoProcessor.from_pretrained("openvla/openvla-7b", trust_remote_code=True)
vla = AutoModelForVision2Seq.from_pretrained(
    "openvla/openvla-7b",
    attn_implementation="flash_attention_2",  # 需要`flash_attn`
    torch_dtype=torch.bfloat16,
    low_cpu_mem_usage=True,
    trust_remote_code=True
).to("cuda:0")

# 获取图像输入并格式化提示
image: Image.Image = get_from_camera(...)  # 假设get_from_camera是获取摄像头图像的函数
prompt = "In: What action should the robot take to {<INSTRUCTION>}?\nOut:"

# 预测动作（7自由度；对BridgeData V2解归一化）
inputs = processor(prompt, image).to("cuda:0", dtype=torch.bfloat16)
action = vla.predict_action(**inputs, unnorm_key="bridge_orig", do_sample=False)

# 执行动作...
robot.act(action, ...)

but，l face the problem as follow
OSError: We couldn't connect to 'https://huggingface.co' to load this file, couldn't find it in the cached files and it looks like openvla/openvla-7b is not the path to a directory containing a file named config.json.
Checkout your internet connection or see how to run the library in offline mode at 'https://huggingface.co/docs/transformers/installation#offline-mode'.

Zeus2E, 2025-05-12:
processor = AutoProcessor.from_pretrained("(the local model filepath)", trust_remote_code=True)

Make sure you have downloaded the complete files from hgface

xyl9q, 2025-05-21:
> processor = AutoProcessor.from_pretrained("(the local model filepath)", trust_remote_code=True)处理器 = AutoProcessor.from_pretrained（“（本地模型文件路径）”， trust_remote_code=True）
> 
> Make sure you have downloaded the complete files from hgface确保您已从 hgface 下载了完整的文件

Thank you for your guidance.

closed 2025-05-21 by xyl9q