openvla/openvla #274: How to run fine-tuned model

Martin-gitman, 2025-07-03:
Hey everyone,

I've successfully fine-tuned the system using LoRA as described in the documentation, adapting it to my custom task and embodiment. Now I’d like to run inference using my fine-tuned model.

I'm a bit stuck on how to modify the inference script accordingly. Here’s what I’ve figured out so far:

I assume I need to load the fine-tuned model from ft_logs.

However, I'm not sure what my unnorm_key should be.

I believe it's related to the dataset_statistics.json, but I'm unclear on how exactly to determine or use the correct key.

```
# Install minimal dependencies (`torch`, `transformers`, `timm`, `tokenizers`, ...)
# > pip install -r https://raw.githubusercontent.com/openvla/openvla/main/requirements-min.txt
from transformers import AutoModelForVision2Seq, AutoProcessor
from PIL import Image

import torch

# Load Processor & VLA
processor = AutoProcessor.from_pretrained("openvla/openvla-7b", trust_remote_code=True)
vla = AutoModelForVision2Seq.from_pretrained(
    "openvla/openvla-7b", 
    attn_implementation="flash_attention_2",  # [Optional] Requires `flash_attn`
    torch_dtype=torch.bfloat16, 
    low_cpu_mem_usage=True, 
    trust_remote_code=True
).to("cuda:0")

# Grab image input & format prompt
image: Image.Image = get_from_camera(...)
prompt = "In: What action should the robot take to {<INSTRUCTION>}?\nOut:"

# Predict Action (7-DoF; un-normalize for BridgeData V2)
inputs = processor(prompt, image).to("cuda:0", dtype=torch.bfloat16)
action = vla.predict_action(**inputs, unnorm_key="bridge_orig", do_sample=False)

# Execute...
robot.act(action, `...)`

```

closed 2025-07-30 by Martin-gitman

krueger-m, 2025-10-23:
Yes :) I can currently not share my entire script with you but
the according code I have is:

path= MODEL_PATH
	processor = AutoProcessor.from_pretrained(path, trust_remote_code=True)
	vla = AutoModelForVision2Seq.from_pretrained(
		path,   #"openvla/openvla-7b",
		#attn_implementation="flash_attention_2",  # [Optional] Requires `flash_attn` 
		torch_dtype=torch.bfloat16,
		low_cpu_mem_usage=True,
		trust_remote_code=True
	).to("cuda:0")

	dataset_statistics_path = os.path.join(path, "dataset_statistics.json")
	if os.path.isfile(dataset_statistics_path):
		with open(dataset_statistics_path, "r") as f:
			norm_stats = json.load(f)
		vla.norm_stats = norm_stats
	else:
		print(
			"WARNING: No local dataset_statistics.json file found for current checkpoint.\n"
			"You can ignore this if you are loading the base VLA (i.e. not fine-tuned) checkpoint."
			"Otherwise, you may run into errors when trying to call `predict_action()` due to an absent `unnorm_key`."
			)

....

action = vla.predict_action(**inputs, unnorm_key="working_dataset", do_sample=False)

For me "working_dataset" is the name of the fine-tuning dataset that I have used for training.
I think I had to add this name into some files before training.

XiaoweiLinXL, 2025-10-24:
Thank you, Martin!