openvla/openvla #281: Cannot reproduce the results on LIBERO

Boltzmachine, 2025-07-17:
I just use the provided finetuned script with the hyperparameters listed in https://huggingface.co/openvla/openvla-7b-finetuned-libero-spatial#openvla-7b-fine-tuned-on-libero-spatial.

Here's the training script

```shell
CUDA_VISIBLE_DEVICES=0,1,2,3 torchrun --standalone --nnodes 1 --nproc-per-node 4 vla-scripts/finetune.py \
  --vla_path "openvla/openvla-7b" \
  --data_root_dir datasets/open-x-embodiment \
  --dataset_name libero_spatial_no_noops \
  --lora_rank 32 \
  --batch_size 6 \
  --grad_accumulation_steps 1 \
  --learning_rate 5e-4 \
  --image_aug True \
  --save_steps 2000 \
  --max_steps 50000
```

And the evaluation script
```
python experiments/robot/libero/run_libero_eval.py  --model_family openvla  --pretrained_checkpoint outputs/openvla-7b+libero_spatial_no_noops+b6+lr-0.0005+lora-r32+dropout-0.0--image_aug  --task_suite_name libero_spatial --center_crop True
```

As a result, I only get a performance of 79.0%, which is lower than the reported 84.7%.

learnerljh, 2025-07-18:
Maybe you should increase batchsize. 你的batchsize和官方超参数不同。我的实验也显示batchsize会影响收敛

Boltzmachine, 2025-07-19:
Thanks! did you reproduce the results on LIBERO by any chance?