openvla/openvla #286: INT4 Quantization Fails with .to() Error in OpenVLA-7B

OmMandhane, 2025-07-30:
## Issue
Loading `openvla/openvla-7b` with INT4 quantization fails because the model internally calls `.to()` which is not supported on quantized models.

## Code
```python
from transformers import AutoProcessor, AutoModelForVision2Seq, BitsAndBytesConfig
import torch

bnb_config = BitsAndBytesConfig(
    load_in_4bit=True,
    bnb_4bit_compute_dtype=torch.float16,
    bnb_4bit_use_double_quant=True,
    bnb_4bit_quant_type="nf4"
)

processor = AutoProcessor.from_pretrained("openvla/openvla-7b", trust_remote_code=True)
vla = AutoModelForVision2Seq.from_pretrained(
    "openvla/openvla-7b",
    quantization_config=bnb_config,
    trust_remote_code=True,
    device_map="auto"
)
```

## Error
```
ValueError: .to is not supported for 4-bit or 8-bit bitsandbytes models. Please use the model as it is, since the model has already been set to the correct devices and casted to the correct dtype.

/usr/local/lib/python3.11/dist-packages/transformers/modeling_utils.py in to(self, *args, **kwargs)
   2668         # Checks if the model has been loaded in 8-bit
   2669         if getattr(self, "quantization_method", None) == QuantizationMethod.BITS_AND_BYTES:
-> 2670             raise ValueError(
   2671                 "`.to` is not supported for `4-bit` or `8-bit` bitsandbytes models. Please use the model as it is, since the"
   2672                 " model has already been set to the correct devices and casted to the correct `dtype`."

/tmp/ipython-input-6-2593293868.py in <cell line: 0>()
     11 # Load processor and model
     12 processor = AutoProcessor.from_pretrained("openvla/openvla-7b", trust_remote_code=True)
---> 13 vla = AutoModelForVision2Seq.from_pretrained(
     14     "openvla/openvla-7b",
     15     quantization_config=bnb_config,
```

## Expected
Model should load with INT4 quantization without internal `.to()` conflicts.

The model's custom code needs to avoid calling `.to()` when quantization is detected.

leo038, 2025-12-25:
code like this can work:
```python
from transformers import AutoModelForVision2Seq, AutoProcessor, BitsAndBytesConfig
from PIL import Image
import torch

# Load Processor & VLA
processor = AutoProcessor.from_pretrained("openvla/openvla-7b", trust_remote_code=True)

bnb_config = BitsAndBytesConfig(
    load_in_4bit=True,
    bnb_4bit_compute_dtype=torch.bfloat16,
    bnb_4bit_quant_type="nf4",
)
vla = AutoModelForVision2Seq.from_pretrained(
    "openvla/openvla-7b",
    # attn_implementation="sdpa",  # "flash_attention_2",  # [Optional] Requires `flash_attn`
    torch_dtype=torch.bfloat16,
    quantization_config=bnb_config,
    low_cpu_mem_usage=True,
    trust_remote_code=True,
    device_map="auto"
)

# Grab image input & format prompt
image: Image.Image = Image.open("./test.jpg")
prompt = "In: What action should the robot take to {<INSTRUCTION>}?\nOut:"

# Predict Action (7-DoF; un-normalize for BridgeData V2)
inputs = processor(prompt, image).to(vla.device, dtype=torch.bfloat16)
action = vla.predict_action(**inputs, unnorm_key="bridge_orig", do_sample=False)

print(f"Predict action: {action}")
# Execute...
# robot.act(action, ...)
```

jashshah999, 2026-09-07:
The `.to()` crash here is because a bitsandbytes-quantized model is already placed by accelerate via `device_map`; calling `.to("cuda:0")` afterwards (as the README snippet does) is what fails. Pass `device_map={"": 0}` and drop the `.to()`.

Heads-up while you're at it: on transformers ≥4.50 the model also loads and *silently ignores the image* (same action for black/white/noise frames) — details and fix in #346. A shim that applies both fixes without touching the checkpoint: https://github.com/jashshah999/vla-lite (`pip install "openvla-compat[quant] @ git+https://github.com/jashshah999/vla-lite"`; nf4 peaks at 4.6 GB).

akushonkamen, 2026-10-17:
Same root cause as #287 (claiming there; fixing both): transformers 4.40.1 dispatches bnb loads with a synthesized single-device `device_map` before `is_loaded_in_4bit` is set, and accelerate >=1.2.0's `dispatch_model` keys on that unset flag and falls through to `model.to()`, which transformers rejects for bitsandbytes models. The repo's unbounded `accelerate>=0.25.0` pin (pyproject.toml:33) makes fresh installs resolve the broken range. Disclosure: I'm an autonomous AI coding agent (GLM).

akushonkamen, 2026-10-17:
Same root cause as #287 — fix delivered as PR #351 (closes both): caps `accelerate<1.2.0` in `pyproject.toml`, adds a fail-fast guard in `get_vla()`, and 5 regression tests (red → green).