openvla/openvla #299: LoRA finetune on Libero Spatial yields 0% success rate

Lastsource, 2025-09-02:
Hello, I am trying to fine-tune OpenVLA on the Libero Spatial dataset following the instructions in the LoRA finetune README. I have converted the dataset as described, and I checked that the dataset_statistics.json file is similar to the “no_noop” version.

I used a batch size of 16 for finetuning. However, when I evaluate the model on the Libero Spatial tasks, the measured success rate is 0%, even though the rollout videos show that the robot moves but fails to complete the tasks.

Steps I followed:
	1.	Converted the Libero Spatial dataset according to the README.
	2.	Verified the dataset_statistics.json.
	3.	Finetuned using LoRA with batch size 16.
	4.	Evaluated using run_libero_eval.py with the checkpoint.

Is there anything I might be missing in the conversion, preprocessing, or LoRA finetuning procedure that could lead to zero success on Libero Spatial? Any guidance would be greatly appreciated!

wanlh99, 2025-09-08:
me too. Has this problem been solved?

gaogehan, 2025-10-22:
Me too, Have you addressed this problem? thank you!

v-xchen-v, 2026-01-08:
Me too. For more details:
1. When evaluating the base model, the eef almost don't move. It make sense since it haven't see Libero-Spatial.
2. When evaluating the lora finetuned model, the eef move a little but not in correct directly and drop to the table after a while. And of course, 0 success rate. 
3. The different betweem my exp and the readme guide, is I decrease batchsize from 16 to 4. But since I used 4 GPU with --nproc-per-node 4, it shouldn't influence a lot. 
4. I trained 8k steps, and the training curve is weird as well:

[image]