openvla/openvla #301: installation failure : prismatic.training

yanan1116, 2025-09-10:
hi authors,

this is an installation failure issue.

 i tried to install with minimum: `pip install -r https://raw.githubusercontent.com/openvla/openvla/main/requirements-min.txt`
then when I try to load the model:
```python
from transformers import AutoModelForVision2Seq, AutoProcessor
from PIL import Image

import torch

# Load Processor & VLA
processor = AutoProcessor.from_pretrained("openvla/openvla-7b", trust_remote_code=True)
vla = AutoModelForVision2Seq.from_pretrained(
    "openvla/openvla-7b", 
    attn_implementation="flash_attention_2",  # [Optional] Requires `flash_attn`
    torch_dtype=torch.bfloat16, 
    low_cpu_mem_usage=True, 
    trust_remote_code=True
).to("cuda:0")
```

here is the error 
```
Traceback (most recent call last):
  File "/home/yanan/robotics/openvla/ttt.py", line 8, in <module>
    vla = AutoModelForVision2Seq.from_pretrained(
  File "/home/yanan/anaconda3/envs/openvlaa/lib/python3.10/site-packages/transformers/models/auto/auto_factory.py", line 550, in from_pretrained
    model_class = get_class_from_dynamic_module(
  File "/home/yanan/anaconda3/envs/openvlaa/lib/python3.10/site-packages/transformers/dynamic_module_utils.py", line 489, in get_class_from_dynamic_module
    final_module = get_cached_module_file(
  File "/home/yanan/anaconda3/envs/openvlaa/lib/python3.10/site-packages/transformers/dynamic_module_utils.py", line 315, in get_cached_module_file
    modules_needed = check_imports(resolved_module_file)
  File "/home/yanan/anaconda3/envs/openvlaa/lib/python3.10/site-packages/transformers/dynamic_module_utils.py", line 180, in check_imports
    raise ImportError(
ImportError: This modeling file requires the following packages that were not found in your environment: prismatic. Run `pip install prismatic`
```

then I executed `pip install prismatic`
```
Collecting prismatic
  Using cached prismatic-0.4-py2.py3-none-any.whl.metadata (586 bytes)
Collecting isodate (from prismatic)
  Using cached isodate-0.7.2-py3-none-any.whl.metadata (11 kB)
Using cached prismatic-0.4-py2.py3-none-any.whl (7.6 kB)
Using cached isodate-0.7.2-py3-none-any.whl (22 kB)
Installing collected packages: isodate, prismatic
Successfully installed isodate-0.7.2 prismatic-0.4
```

then I executed `pip show prismatic`
```
Name: prismatic
Version: 0.4
Summary: Serialization between JSON dicts and data objects designed for REST APIs
Home-page: https://github.com/adambyrtek/prismatic
Author: Adam Byrtek
Author-email: adambyrtek@gmail.com
License: MIT
Location: /home/yanan/anaconda3/envs/openvlaa/lib/python3.10/site-packages
Requires: isodate
Required-by
```

however, when I run the model loading code again, the error still exists.

any solution ?

yanan1116, 2025-09-10:
then I also tried using `pip install -e .` manner.
but it still does not work, with the error message:
```
Traceback (most recent call last):
  File "/home/yanan/robotics/openvla/ttt.py", line 8, in <module>
    vla = AutoModelForVision2Seq.from_pretrained(
  File "/home/yanan/anaconda3/envs/openvla/lib/python3.10/site-packages/transformers/models/auto/auto_factory.py", line 550, in from_pretrained
    model_class = get_class_from_dynamic_module(
  File "/home/yanan/anaconda3/envs/openvla/lib/python3.10/site-packages/transformers/dynamic_module_utils.py", line 501, in get_class_from_dynamic_module
    return get_class_in_module(class_name, final_module)
  File "/home/yanan/anaconda3/envs/openvla/lib/python3.10/site-packages/transformers/dynamic_module_utils.py", line 201, in get_class_in_module
    module = importlib.machinery.SourceFileLoader(name, module_path).load_module()
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/yanan/.cache/huggingface/modules/transformers_modules/openvla/openvla-7b/31f090d05236101ebfc381b61c674dd4746d4ce0/modeling_prismatic.py", line 24, in <module>
    from prismatic.training.train_utils import (
ModuleNotFoundError: No module named 'prismatic.training.train_utils'
```

besides, i have not found anything related to `prismatic.training.train_utils` under this path: https://github.com/openvla/openvla/tree/main/prismatic/training

lovecode1, 2025-09-17:
Did you check for such a file under `/home/yanan/.cache/huggingface/modules/transformers_modules/openvla/openvla-7b/31f090d05236101ebfc381b61c674dd4746d4ce0/`? Looks like it's code that came from huggingface?

yanan1116, 2025-09-17:
@lovecode1  yes there is. 

should I delete this folder ?

```
/home/yanan/.cache/huggingface/modules/transformers_modules/openvla/openvla-7b/31f090d05236101ebfc381b61c674dd4746d4ce0/
├── configuration_prismatic.py
├── __init__.py
├── modeling_prismatic.py
├── processing_prismatic.py
└── __pycache__
    ├── configuration_prismatic.cpython-310.pyc
    ├── configuration_prismatic.cpython-311.pyc
    ├── configuration_prismatic.cpython-312.pyc
    ├── modeling_prismatic.cpython-310.pyc
    ├── modeling_prismatic.cpython-311.pyc
    ├── processing_prismatic.cpython-310.pyc
    ├── processing_prismatic.cpython-311.pyc
    └── processing_prismatic.cpython-312.pyc
```

lovecode1, 2025-09-18:
Try using Gemini code cli to troubleshoot it. Most likely it's a simple fix.

jianbinglin128-alt, 2026-01-20:
Hello, how did you solve this problem？